ROBOTIS-GIT/OpenCR #163: Expose firmwave version information

LucidOne, 2019-02-21:
It would be useful to expose the OpenCR firmware version in device descriptor's iProduct or configuration descriptor''s iConfiguration, even when the device is in usb to serial passthrough mode.

This will make it easier to diagnose when the firmware needs to be upgraded and that any upgrades have been successful.

LucidOne, 2019-02-24:
Also OpenCR firmware should be versioned with [SemVer](https://semver.org/)

OpusK, 2019-02-25:
Hi, @LucidOne. 

Thank you for your feedback.
I totally agree with your feedback on version control.
In this regard, we will discuss how to apply it to our team.

OpusK, 2019-02-25:
> It would be useful to expose the OpenCR firmware version in device descriptor's iProduct or configuration descriptor''s iConfiguration, even when the device is in usb to serial passthrough mode.

Regarding version control for OpenCR drivers, you can do it the way you said.
However, in this case, because the information is different from the boot loader information, there will be a situation where the port is changed each time the boot loader and application are moved.
Therefore, it will be difficult to officially support it.

Of course, if you modify it to suit your project, you will be able to use this function enough.

LucidOne, 2019-02-25:
Just so we are on the same page, we are suggesting changing 
`iProduct` to `OpenCR Virtual ComPort in FS Mode v1.4.5`
or
`iSerial` to `1405DEADBEEF` where `DEADBEEF` is a unique random value possibly derived from the [device id](https://techoverflow.net/2015/02/03/reading-the-stm32-unique-device-id-in-c/)

```
#  lsusb -v -d 0483:5740

Bus 002 Device 004: ID 0483:5740 STMicroelectronics STM32F407
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x5740 STM32F407
  bcdDevice            2.00
  iManufacturer           1 ROBOTIS
  iProduct                2 OpenCR Virtual ComPort in FS Mode
  iSerial                 3 FFFFFFFEFFFF
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           67
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered
```

OpusK, 2019-02-27:
> I can write a pull request for `usb_to_dxl.ino` as an example if that helps.

Would you like to write a full request? There are some things to check for with the Bootloader, so if you write down what you think about it, it will help us consider this part.

Thanks again for your feedback & contribution.

OpusK, 2020-03-06:
It was decided that development/support would be difficult on this issue.

closed 2020-03-06 by OpusK