ROBOTIS-GIT/OpenCR #165: dfu-util de-dup

LucidOne, 2019-02-25:
Are there any OpenCR specific changes to `OpenCR/arduino/opencr_arduino/tools/opencr_tools_1.0.0/linux64/dfu-util`?

```
$ ./dfu-util --version
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org
```

Does it make more sense to use the version of `dfu-util` from the OS package manager or otherwise include it via a git submodule?

Ubuntu 16.04
https://packages.ubuntu.com/xenial/dfu-util

macOS
https://formulae.brew.sh/formula/dfu-util

Windows
https://chocolatey.org/packages/dfu-util

It looks like there is 3 copies of the same dfu-util binary for linux and macos, while only 1 for windows.
```
$ find . -type f -name dfu-util | xargs md5sum
cf3fb546593a03e29a4f8c2cb3222cbf  ./arduino/opencr_develop/opencr_fw_template/opencr_fw/tools/linux64/dfu-util/dfu-util
412bfaa0e6fb631dafc6bc1f2d419066  ./arduino/opencr_develop/opencr_fw_template/opencr_fw/tools/macosx/dfu-util/dfu-util
4feda8df892008e50c15a0f041d81ea7  ./arduino/opencr_develop/opencr_fw_template/opencr_fw/tools/linux/dfu-util/dfu-util
cf3fb546593a03e29a4f8c2cb3222cbf  ./arduino/opencr_develop/opencr_fw_template/opencr_fw_arduino/tools/linux64/dfu-util/dfu-util
412bfaa0e6fb631dafc6bc1f2d419066  ./arduino/opencr_develop/opencr_fw_template/opencr_fw_arduino/tools/macosx/dfu-util/dfu-util
4feda8df892008e50c15a0f041d81ea7  ./arduino/opencr_develop/opencr_fw_template/opencr_fw_arduino/tools/linux/dfu-util/dfu-util
cf3fb546593a03e29a4f8c2cb3222cbf  ./arduino/opencr_arduino/tools/opencr_tools_1.0.0/linux64/dfu-util/dfu-util
412bfaa0e6fb631dafc6bc1f2d419066  ./arduino/opencr_arduino/tools/opencr_tools_1.0.0/macosx/dfu-util/dfu-util
4feda8df892008e50c15a0f041d81ea7  ./arduino/opencr_arduino/tools/opencr_tools_1.0.0/linux/dfu-util/dfu-util
$ find . -type f -name dfu-util.exe | xargs md5sum
640a8df448e1dbe1b738d138e83da483  ./arduino/opencr_arduino/tools/opencr_tools_1.0.0/win/dfu-util/dfu-util.exe
```

As far as I can tell, at minimum these directories should be removed
```
./arduino/opencr_develop/opencr_fw_template/opencr_fw/tools
./arduino/opencr_develop/opencr_fw_template/opencr_fw_arduino/tools
```
I'm not sure this directory belongs in the OpenCR repository, but at the very least this should have a README that clarifies if they have been modified from upstream and a link to the source code.
```
./arduino/opencr_arduino/tools/opencr_tools_1.0.0/
```

OpusK, 2019-02-25:
Hi, @LucidOne 

> Does it make more sense to use the version of `dfu-util` from the OS package manager or otherwise include it via a git submodule?

Arduino downloads these tools from the json file as shown below.
- [json file](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_release/package_opencr_index.json#L1042)

As you said, we could manage by submodule and create a separate compressed file like this to maintain the folder structure of OpenCR tools.
In this regard, I will discuss with the team related this issue.

> It looks like there is 3 copies of the same dfu-util binary for linux and macos, while only 1 for windows.

[opencr_fw_template](https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_develop/opencr_fw_template) directory is used for other purposes apart from Arduino. I will discuss with the person in charge.

LucidOne, 2019-02-25:
I think some of my confusion is that several releases are being generated from one source tree.
https://github.com/ROBOTIS-GIT/OpenCR/releases/tag/1.0.0
https://github.com/ROBOTIS-GIT/OpenCR/releases/tag/1.4.5
Since `opencr_core` and `opencr_tools` are not being released at the same time, maybe they should be versioned in separate repos.

Having a few extra `README.md` files to describe the purpose of the sub-directories would help.

OpusK, 2019-02-27:
I updated README.

And [I deleted fw_template related files.](https://github.com/ROBOTIS-GIT/OpenCR/commit/554ce4a1764baf991f94d8ba9a81877935f35bbd) It will be merged later as a master.

closed 2019-03-19 by OpusK