ROBOTIS-GIT/OpenCR #167: opencr_ld de-dup

LucidOne, 2019-02-25:
This is somewhat related to https://github.com/ROBOTIS-GIT/XelNetwork_Manager/issues/1

How is `opencr_ld` different than `xel_loader` and `opencm9.04_ld`?
Why do all of these boards need separate tools?

OpenCM9.04 STM32F103CB
OpenCR STM32F746ZGT6
SensorXel STM32L151C8T6-A
PowerXel STM32L151C8T6-A
CommXel STM32F746ZGT6

OpusK, 2019-02-25:
> Why do all of these boards need separate tools?

This is because the protocols of OpenCM 9.04 and OpenCR and XEL Network bootloaders are different.
Therefore, the loader is bound to be different.

Note that all boards in the XEL Network use the same loader. (These are using the same boot loader protocol.)

LucidOne, 2019-02-25:
It looks like the OpenCR board and the CommXel use the same processor, so it seems to me that it might be less maintenance to use the same bootloaders.

I'm wondering if at some point in the future all of the Robotis STM32 based products could use the same bootloader to simplify the documentation and hardware setup issues for end users.

OpusK, 2019-02-25:
Older products (OpenCR, OpenCM) are having a lot of trouble changing the boot loader due to compatibility issues with previously sold products.
Therefore, it is difficult to modify OpenCR and OpenCM boot loaders.

However, from the XEL Network onward, we are going to proceed with the same boot loader protocol.

LucidOne, 2019-02-25:
That sounds great. Robotis has put an enormous amount of work into writing software and documentation so some duplication and [technical debt](https://en.wikipedia.org/wiki/Technical_debt) is to be expected. I appreciate your help with the confusing parts as I try to read all of it! :slightly_smiling_face:

closed 2019-02-27 by OpusK