ROBOTIS-GIT/OpenCR #170: Can't connect  XL-430 after Firmware Recovery Mode

EndlessLoops, 2019-03-19:
**hi，
A few day ago ,i upload the source code of OpenManipulator with tb3 but an error occurred ：**

```
Arduino: 1.8.9 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

Sketch uses 181896 bytes (23%) of program storage space. Maximum is 786432 bytes.
Global variables use 46364 bytes of dynamic memory.
opencr_ld ver 1.0.2
An error occurred while uploading the sketch
opencr_ld_main
>>
file name : /tmp/arduino_build_375437/turtlebot3_core.ino.bin
file size : 179 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
cmd_read_board_name fail : 0xF020
```

**then i forced to enter "firmware recovery mode" after seeing the[ issue ](https://github.com/ROBOTIS-GIT/OpenCR/issues/150)，**

```
lalala@lalala-Inspiron-3576:~/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0/linux64$ ll
total 228
drwxrwxr-x 3 lalala lalala   4096 3月  18 17:11 ./
drwxrwxr-x 6 lalala lalala   4096 3月  18 15:38 ../
drwxrwxr-x 2 lalala lalala   4096 10月 19  2016 dfu-util/
-rwxrwxr-x 1 lalala lalala 183440 3月  18 16:54 opencrfw.bin*
-rwxrwxr-x 1 lalala lalala  34416 10月 19  2016 opencr_ld*
lalala@lalala-Inspiron-3576:~/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0/linux64$ sudo ./opencr_ld /dev/ttyACM0 115200 ./opencrfw.bin 1
opencr_ld ver 1.0.2
opencr_ld_main
>>
file name : ./opencrfw.bin
file size : 179 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 0.905000 sec
flash_write : 0 : 1.692000 sec
CRC OK 123CEA6 123CEA6 0.006000 sec
[OK] Download

```

**the i reload the source code of tb3_burger in the OpenCR**   

```
Sketch uses 181896 bytes (23%) of program storage space. Maximum is 786432 bytes.
Global variables use 46364 bytes of dynamic memory.
opencr_ld ver 1.0.2
opencr_ld_main
>>
file name : /tmp/arduino_build_217208/turtlebot3_core.ino.bin
file size : 179 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 0.973000 sec
flash_write : 0 : 1.689000 sec
CRC OK 1254208 1254208 0.006000 sec
[OK] Download
jump_to_fw
```

**but when i try to test by PUSH SW1 and SW2 ,the XL-430 motors can't work .  When i executed the bringup command ，the terminal showed the normal messages without error . but the burger can't move by using Keyboard controling.  

The XL-430 motors and cable have been used normally .

 I am a newbie. How should I find out the cause of the error and fix it?

I look forward to and sincerely thank you for your reply.**

unverciftci, 2019-04-14:
I have the same issue. Help is needed,please.

RyanJaehyunShim, 2019-04-24:
@EndlessLoops @unverciftci 

Would you follow tutorials in our e-manual and let me know if you are stuck? Please refer to the link below.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/setup/#setup)

Regards,
Ryan

RyanJaehyunShim, 2019-05-16:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if you have any other questions. Thanks.

closed 2019-05-16 by RyanJaehyunShim