ROBOTIS-GIT/OpenCR #173: Cannot operate dynamixels

unverciftci, 2019-04-22:
Hi,
OpenCR's led example is working but I cannot operate dynamixels. ID's are correct. I receive no errors but Dynamixel XL4230 doesn't respond.

OpusK, 2019-04-23:
> OpenCR's led example is working but I cannot operate dynamixels. ID's are correct. I receive no errors but Dynamixel XL4230 doesn't respond.

Hi, @unverciftci 

What is the example you used? Since OpenCR has the DYNAMIXEL Power Enable pin, you must activate it. Have you tested with examples related to DYNAMIXEL? If so, which example did you test?

unverciftci, 2019-05-02:
I tried the position example with power enabled but it doesn't work.

OpusK, 2019-05-03:
Hi, @unverciftci 

Please try to control it from PC through [usb_to_dxl example](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino).

As a precaution, make the [`DXL_BAUD 'value on line 10](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino#L10) equal to the speed of the actuator you want to use.

mclaubee, 2019-05-28:
Hi, I was having the same problem. Everything is working except DXL port.

OpusK, 2019-05-29:
Hi, @mclaubee 

What example are you using?
And, did you put the proper power for OpenCR?
If you have a multimeter, can you check if the voltage is normal?

And would you check if the switch in the picture below is on?

[image]

mclaubee, 2019-05-29:
​Hi Kai, 
Thanks for your prompt reply. Yes, I use a 12V power supply, from OP3. And yes I turn the switch on. 

I tried opencr_op3, it couldn't turn on the dynamixel's port power (24), returned an unknown error. 

Then I tried ping, model_scan, Find_Dynamixel, usb_to_dxl, etc from Arduino, but none of them is able to find the servos.

benoit-girard, 2019-07-16:
I have a similar problem: using a turtlebot3, I followed the OpenCR setup instructions (http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup), I obtained the expected `jump_to_fw` output, however the "6.3.2 basic operation" test does not work.

The robot does not move after pressing SW1 or SW2, when it operates on the battery (switch on the OpenCR is on).

What could I do to identify the source of the problem?

benoit-girard, 2019-07-16:
Problem solved, but I don't know why: I did again the setup using the raspberry & the shell setup, and now it works. Sorry for the inconvenience.

OpusK, 2019-07-22:
Hi, @benoit-girard 

> I obtained the expected jump_to_fw output, however the "6.3.2 basic operation" test does not work.

This is related to #191 and has been fixed in the Master version.
Since Turtlebot3 core images are managed separately, we recommend using a possible script method.
We apologize for any inconvenience.

OpusK, 2019-07-22:
@mclaubee 

Sorry. I was missing this issue.
Have you been resolved?

OpusK, 2019-07-30:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-07-30 by OpusK

mclaubee, 2019-07-31:
Hi Kei,

No solution yet. I actually just bypassed the OpenCRand directly control the servos through U2D2, just missing the IMU and buttons. But I'm still interested to figure out what's the problem. I will be going to Changwon, Korea in August for FIRA. Will Robotis setting a booth at the venue?