ROBOTIS-GIT/OpenCR #175: Problems controlling the motors and getting odometry at the same time

Sarah4292, 2019-04-23:
Hello, 
I'M a newbie on this field. I have a Problem with my turtlebot3 mecanum. Bevor the SSD Card in the Intel-Nuc from the Robot I used all the time got broken, everthing worked fine. I could control the motors (Dynamixel XM430-W210-T) and read the odometry. I have replaced the SSD card with another one on which is the same software (from another turtlebot 3 mecanum). The Arduino code is the same. But now I can either control the motors and get no odometry or the motors are not controlled and the odometry comes regularly with the value zero.

[turtlebot3_mecanum_core.zip](https://github.com/ROBOTIS-GIT/OpenCR/files/3106580/turtlebot3_mecanum_core.zip)

Does anyone know what this could be? The Arduino code is completely the same as before. 

Thanks for your help. 

Sarah

RyanJaehyunShim, 2019-04-24:
@Sarah4292 

Would you clarify your question again? If you would like to know how to set up a robot pc, please refer to our e-manual (http://emanual.robotis.com/docs/en/platform/turtlebot3/setup/#setup)

Regards,
Ryan

Sarah4292, 2019-04-24:
My Problem is that everything worked fine until I have changed the SSD card from my remote PC. On the new SSD card I've inserted, I have already done all Setups. When I'm uploading the arduino code on the Open CR board there is no errror. When I start publishing velocity commands with ROS, the motors starts and the wheels of to robot are rolling but I do not recieve odometry feedback (rostopic hz /odom_combined). Now I really do not know what the problem is, because I didn't change any code and I get no errors. Sometimes I recive some odometry messages, but then they come irregular.

RyanJaehyunShim, 2019-04-25:
@Sarah4292 

There is no solution I can think of for now. I suggest you try it again with another computer (because you changed your SSD card in your remote PC, the problem might exist in your laptop rather than on the Turtlebot side) and see if it works. If yes, next thing you would want to do is to reset all of your PC and router settings (including formatting your computer and installing Ubuntu and ROS on it) and follow our tutorial again.

If you are suspicious of communication between your robot and computer, you can also refer to the following link (https://www.wireshark.org/download.html).

Regards,
Ryan

RyanJaehyunShim, 2019-05-16:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if you have any other questions. Thanks.

closed 2019-05-16 by RyanJaehyunShim