ROBOTIS-GIT/OpenCR #176: tb3 mechanum with same functionalities as burger or waffle

14nrv, 2019-04-27:
Hello,

I would like to "build" a turtlebot3 mechanum (with an open manipulator after) that can have the same functionalities as the burger or waffle.
For now, I use `arduino > Examples > turtlebot3 > turtlebot3_friends > turtlebot3_mechanum` and it looks like that is only for remote control.
Do you know a way how to have the same functionalities as the burger or waffle?

RyanJaehyunShim, 2019-04-30:
@14nrv 

There is no official ROS package set for mecanum, so we cannot give any help with it, but somebody made it and shared it on his github page. You would want to refer to the link below. 
(https://github.com/Seunghooon/turtlebot3_mecanum)

You will be able to do manipulation + mecanum by combining the ROS package in the above-mentioned link and turtlebot3_with_open_manipulator.

Regards,
Ryan

14nrv, 2019-04-30:
@RyanJaehyunShim 

thx for this very useful link :)

RyanJaehyunShim, 2019-04-30:
@14nrv 

I will close this issue. If you have any other questions, please open another issue.

Good luck with your work,
Ryan

closed 2019-04-30 by RyanJaehyunShim

14nrv, 2021-01-25:
@JaehyunShim, this repo (https://github.com/Seunghooon/turtlebot3_mecanum) doesn't work (I already opened an issue)
Do you know another repo? 😄

ROBOTIS-Will, 2021-01-26:
Hi @14nrv 
The aforementioned repository does not belong to ROBOTIS.
Please contact the repository owner via issue thread as you already did.
Thank you.