ROBOTIS-GIT/OpenCR #177: [Segway - Friends] - Setup error

CesMak, 2019-05-01:
Hey there, 

I wanted to test segway friends arduino code on opencr 1.0 Board. However I used two motors of type XH540-W150-R.

First I configured both motors with 3M. BaudRate and one with ID 1 The other one with ID 2. I connected them in a line and only once cable to the OpenCR board. 

Next I connected USB to pc and I also connected the power to the black power jack. 

I also changed the code marginally (Baudrate to 3M) 

Now I flashed the code onto the OpenCR board....

Then no motor turned.... any ideas why?
Is there something I forgot?

If I open serial monitor I get: 
```
Inside setup:

Inside motor driver init

Port Handler setup

Packet Handler setup

Port is opened
Baudrate is set
Inside setTorque motor_driver.cpp
71
Inside setTorque motor_driver.cpp
71
Enabled Dynamixel Torque on both wheels
Setup GroupSyncWrite
Inside imu init
Inside imu init -> update while loop
Inside startDynamixelControlInterrupt
Inside control Segway
 P : 4000.00 I : 2.00 D : 78.00 offset : 0.00 output : 0.00 angle : 0.00
 P : 4000.00 I : 2.00 D : 78.00 offset : 0.00 output : 0.00 angle : 0.00
 P : 4000.00 I : 2.00 D : 78.00 offset : -0.01 output : 0.00 angle : 0.00
 P : 4000.00 I : 2.10 D : 78.00 offset : -0.01 output : 0.00 angle : 0.00
 P : 4000.00 I : 2.10 D : 78.00 offset : -0.01 output : 0.00 angle : 0.00
 P : 4000.00 I : 2.10 D : 78.00 offset : -0.01 output : 0.00 angle : 0.00
 P : 4000.00 I : 2.10 D : 78.00 offset : -0.01 output : 0.00 angle : 0.00
 P : 4000.00 I : 2.10 D : 78.00 offset : -0.02 output : 0.00 angle : 0.00
....

```

RyanJaehyunShim, 2019-05-02:
@CesMak 

If you check the turtlebot3_segway_motor_driver.h file, you will find a list of control table addresses (which is about XM430 motors). 

Please, check out the following link (with control table addresses of XH540) and change the above-mentioned part accordingly. 
(http://support.robotis.com/ko/product/actuator/dynamixel_x/xh_series/xh540-v150.htm)

If you have XM430 motors, I recommend you test your code with them first.

Regards,
Ryan

CesMak, 2019-05-02:
Hey, 

yeah you mean these here?: 

```
// Control table address (Dynamixel X-series)
// for Dynamixel XM430-W210-T http://support.robotis.com/en/product/actuator/dynamixel_x/xm_series/xm430-w210.htm
// for Dynamixel XH540-W15-R http://emanual.robotis.com/docs/en/dxl/x/xh540-w150/#control-table-of-eeprom-area
// same addresses
#define ADDR_X_TORQUE_ENABLE            64
#define ADDR_X_GOAL_PWM                 100
#define ADDR_X_GOAL_VELOCITY            104
#define ADDR_X_PROFILE_ACCELERATION     108
#define ADDR_X_PROFILE_VELOCITY         112
#define ADDR_X_GOAL_POSITION            116
#define ADDR_X_REALTIME_TICK            120
#define ADDR_X_PRESENT_VELOCITY         128
#define ADDR_X_PRESENT_POSITION         132

// Limit values (XM430-W210-T)
#define LIMIT_X_MAX_VELOCITY            240

// Data Byte Length
#define LEN_X_TORQUE_ENABLE             1
#define LEN_X_GOAL_PWM                  2
#define LEN_X_GOAL_VELOCITY             4
#define LEN_X_GOAL_POSITION             4
#define LEN_X_REALTIME_TICK             2
#define LEN_X_PRESENT_VELOCITY          4
#define LEN_X_PRESENT_POSITION          4
```

Well the first addresses are the same for X Series:

- XM430-W210-T
- XH540-W15-R

and the Data Byte Length is also the same or? 

Except the `#define LIMIT_X_MAX_VELOCITY            240` might be different but I do not find the number here:
http://emanual.robotis.com/docs/en/dxl/x/xh540-w150/#control-table-of-eeprom-area

RyanJaehyunShim, 2019-05-03:
@CesMak 

Would you check if your motors are set to PWM mode? You possibly missed our instruction we wrote on NOTE section. 
(http://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#turtlebot3-friends-segway)

If you would like to know how to, please check out the link below for the software you need to use to check the motor mode.
(http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/)

Regards,
Ryan

CesMak, 2019-05-03:
Ah yeah:

NOTE:

    Two Dynamixel X 430 Series are need to be set on PWM Mode. 

I did not see that..... 

XH-540-W150 is not supported for dynamixel_wizard2 is it??? 

I use R+ software to change. So I guess I have to change:
Operating Mode(11) to 

16 | PWM Control Mode  (Voltage Control Mode) | This mode directly controls PWM output. (Voltage Control Mode)
-- | -- | --

RyanJaehyunShim, 2019-05-03:
@CesMak 

I think so... whichever tool you use, try it and let me know if it works.

Ryan

CesMak, 2019-05-03:
Hey,

yap changing to PWM now the motors are driving however I changed:
PWM_Limit in segway.h I changed it from 885 to 36
as this limit of XH-540-W150 is 36. With this limit the motors rotate very slowly....
I changed it back to 885 and now the motors rotate quite fast (as they should). 

However I am not sure if I do damage the motor if I set it back to 885 .....
Can I set this limit independently from the motor specs?

RyanJaehyunShim, 2019-05-03:
@CesMak 

36 is not the maximum PWM limit of the motor but it is the address of the PWM limit, so you can set it up to 885 which is the actual maximum of PWM limit. 

Please create a new issue if you have any other questions, and good luck with your work,
Ryan

closed 2019-05-03 by RyanJaehyunShim