ROBOTIS-GIT/OpenCR #181: Connect FSR to OP3

VasilyevGeorgy, 2019-05-14:
Hi!
I have tried connect OP2's FSR to OP3's OpenCR
Thus I have added FSR as sensors in OP3.robot file, and add some lines of code to open_cr_module related to data read from FSR
After all this manipulations error occurs when I try to launch op3_manager because of wrong protocol type in order to work with FSR

Is there some way to get data from FSR by means Arduino UNO?

Regards,
George

ROBOTIS-Kayman, 2019-06-12:
Hi,

I'm sorry to reply late.
OP2's FSR works with protocol 1.0 and TTL communication. It does not fit OP3.
This is because OP3 uses protocol 2.0 and RS485 communication.
I don't know how to get the data using Arduino.

Regards,
Kayman

VasilyevGeorgy, 2019-06-13:
So, we have solved this problem by using the OpenCM as additional sub controller to get FSR data and toss it through USB to mainboard.
But the new issue has occurred here. To test FSR by means RoboPlus  we have pull out U2D2 from ROBOTIS-OP3. After this all cables and U2D2 were connected according to the their previous position. Intel NUC works fine, but when I try to launch op3_manager it fails with an error: 
```
[ERROR] [1532100652.759606870]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.895751751]: Fail to control LED [[RxPacketError] Unknown error code!] 
```
which is same for this issue: [ https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/issues/50](url)

We have checked OpenCR electronic components, they are ok, although there is no 12V supply voltage on pins. Following your advice from above mentioned issue, we have uploaded OP3 firmware for OpenCR. 
However there is still no 12V to power up servos and previous error occurs.

What could cause this behavior of OpenCR?

Regards,
George

OpusK, 2019-07-30:
Hi, @VasilyevGeorgy 

Sorry, I was missing this issue, and the answer was late.

> We have checked OpenCR electronic components, they are ok, although there is no 12V supply voltage on pins. Following your advice from above mentioned issue, we have uploaded OP3 firmware for OpenCR.
However there is still no 12V to power up servos and previous error occurs.

OpenCR is connected to a GPIO that can disconnect the 12V power to the DXL port.
It turns the power on / off by software (firmware) command.
If the firmware stops working or does not perform the power enable operation, the 12V power may not be turned on.

I recommend uploading the usb_to_dxl example to OpenCR and checking to see if the 12V voltage is normal on the DXL port.

VasilyevGeorgy, 2019-08-08:
Thanks for your help!
We have tried to check controller with RoboPlus and additional software from ROBOTIS, but there was no any response from OpenCR.
We found the problem:  NC7WZ241K8X element was broken and we have unsoldered it.
After this op3_manager starts fine but new error has occurred:
`*** Error in /home/robotis/catkin_ws/devel/lib/op3_manager/op3_manager': free(): invalid next size (fast): 0x00007ff93c002920 ***`

Regards,
George

OpusK, 2019-08-09:
@VasilyevGeorgy 

> We have tried to check controller with RoboPlus and additional software from ROBOTIS, but there was no any response from OpenCR.

Originally, RoboPlus cannot read OpenCR.
What firmware do you currently upload to OpenCR?

> We found the problem: NC7WZ241K8X element was broken and we have unsoldered it.

If you only use RS485 (4pin), no problem.

> After this op3_manager starts fine but new error has occurred:
*** Error in /home/robotis/catkin_ws/devel/lib/op3_manager/op3_manager': free(): invalid next size (fast): 0x00007ff93c002920 ***

I don't know about package-related details, so please leave issues in the OP3 package repository.

OpusK, 2019-08-29:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-08-29 by OpusK

eattooth, 2020-01-29:
Did you succeed in the end?