ROBOTIS-GIT/OpenCR #183: OpenCR board cant find Dynamixel

HaShaq, 2019-06-09:
Hi,
iam currently working on the OpenManipulator 6DOF Sara Version. Iam using the OpenCR Board to control it. Everything was working fine till recently. Iam using this config:

- Successfully uploaded USB_to_DXL to the board
- 12V extern SMPS voltage is used
- Power switch is turned ON
- Dynamixel Baudrate are set correctly

But recently it stopped working. I cant even find the Dynamixel XM430-210T Servos with the RoboPlus 2.0. The LEDs are blinking, so the board is searching for them, but cant find any. 
The TTL connector is providing the correct 12V voltage and a TTL signal.
Other example programs (from the Arduino IDE) are working fine and the board seems alright. Its just that it cant find any Dynamixel. Do you guys have any ideas?

OpusK, 2019-06-12:
Hi, @HaShaq 

OpenCR provides USB_to_DXL as an example but does not guarantee stability.
If you have a product like U2D2, please try it.

If there is no alternative but OpenCR, I recommend using [Dynamixel Wizard2.0](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2).

Pallav1299, 2019-06-16:
If you are on LINUX, you can use this for detecting the dynamixel:
https://github.com/clebercoutof/mixcell

OpusK, 2019-07-11:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-07-11 by OpusK

Matt93-00, 2021-12-23:
Hi I am using an openCr with a mx64at and a mx106. Firstly I uploaded them to the protocol 2.0. Then I used them with and arduino uno wifi rev 2 and the dynamixel shield. Right now I would like to use it with a portenta h7 and mkr dynamixel shield. The problem is that today....suddenly, the open Cr doesn't find any dynamixel through wizard 2.0 or R+ manager. The 12 Volt like input is good, as the previous upload of dxl_to_usb code. I have really  no idea. To be sure, I scanned them with all id and all protocols but nothing happen. I can only say that today I also uploaded the dynamex wizard 2.0. In the end, also the dynamexels that were just paired, are no more found from the software. And no more working. If I run the code position_mode I can only see the output in serial monitor that is always 0 because the dynamixel doesn't move.
Can you help me?

ROBOTIS-Will, 2021-12-27:
Hi @Matt93-00 

Please check if the LED on DYNAMIXEL flickers once when powered up.
If it doesn't, the fuse on the back of the OpenCR could be blown up.
Thanks!

Matt93-00, 2021-12-27:
Hi @ROBOTIS-Will   I can drive dynamixel using an arduino uno and a dynamixel shield. The system works well also when I use the openCr and I run usb_to_dxl indeed the led turned on once when it has to do it. The problem is that if I run Find_dynamixel or scan_dynamixel the result is empty, like when I try to scan it with dynamixel wizard 2.0 (scanning all protocols and all ID). Where could be the the problem? Think about that also the openCr works well. I have no idea

ROBOTIS-Will, 2021-12-28:
Hi @Matt93-00 
When connecting DYNAMIXEL directly to the OpenCR, you cannot use the examples in the DYNAMIXELShield or DYNAMIXEL2Arduino library, but have to use the OpenCR package examples as below.

[image]

[image]

Matt93-00, 2021-12-28:
hey @ROBOTIS-Will  goodmorning when I said Find_dynamixel code or Scan_dynamixel code, I mean that ones associated to the openCR (inside workbench)

Matt93-00, 2021-12-28:
Today, suddenly, it found my dynamixel. Do you think that there is something wrong on the OpenCR1.0? How can I notice it? I mean that the openCr can run all except with the dynamixel at this point