ROBOTIS-GIT/OpenCR #184: ros2arduino.h no file or directory

giovanidiniz, 2019-06-12:
Hey guys, I began porting my code base to ROS2 and I was hoping to do the same with my robot's firmware but I can't compile even the basic turtlebot3_core from ros2 examples.
I'm trying to use the Arduino IDE but keep getting the same error that ros2arduino.h is not found.
Is there anyway to do this?

OpusK, 2019-06-12:
Hi, @giovanidiniz 

You can install ros2arduino from the Library Manager of the Arduino IDE.
At the next OpenCR release, we will update the TB3 core example to display additional error messages for ros2arduino.

giovanidiniz, 2019-06-12:
Oh man, I feel so dumb now...ugh
Thank you!
Probably question for another post but I'm having trouble installing the Micro XRCE-DDS agent and client following the tutorial on the e-manual
I think I read somewhere that there is an issue with the FastRTPS versions used by the standard ROS2 installation and the one needed by Micro XRCE-DDS
Do you have info on that?

OpusK, 2019-06-12:
@giovanidiniz 

> Probably question for another post but I'm having trouble installing the Micro XRCE-DDS agent and client following the tutorial on the e-manual
I think I read somewhere that there is an issue with the FastRTPS versions used by the standard ROS2 installation and the one needed by Micro XRCE-DDS

Agent is FastRTPS dependent. And ROS2 uses rmw_fastrtps as one of the `RosMiddleWare` (default setting).
For information on this, refer to eProsima's Repository.

Anyway, the tutorial in the current eManual is a description of the ROS2 Crystal version.
Crystal Patch3 is also available, but not for later versions. Regarding support for the ROS2 Dashing version, we will support Dashing(including updating eManual) after it is released, because eProsima is currently preparing to release Agent 1.1.0.

In addition, because TB3 uses ros2arduino, you should use the version of Agent and ROS2 according to the version of ros2arduino you are using.
Information on this can be found in the [README in the ros2arduino](https://github.com/ROBOTIS-GIT/ros2arduino#version-specific-dependencies) repository.

giovanidiniz, 2019-06-13:
Hi Kei,
Thank you so much for the info. Indeed that worked out. 
I'm in a bit of a loss now, though. I need Crystal distro to install the MicroXRCE agent/client but I canonly build the package according to the manual on Dashing. 
When I use Crystal I keep getting error:

`error: ‘rclcpp::QoS Initialization’ has not been declared auto qos = rclcpp::QoS(rclcpp::QoS Initialization::from rmw(custom_qos));`

during building. 
This bug is in the gazebo camera_publisher.cpp file. I tried installing and resinstalling rclcp, since there seems to be a disconnect between the version the file was written for and the one installed with Crystal.

Am I doing something wrong there?

OpusK, 2019-07-11:
> This bug is in the gazebo camera_publisher.cpp file. I tried installing and resinstalling rclcp, since there seems to be a disconnect between the version the file was written for and the one installed with Crystal.

Since this is a dependency problem for ros2's packages, I think it's hard to answer here.

Because of this dependency problem, I am using binary to install.

closed 2019-07-17 by OpusK