ROBOTIS-GIT/OpenCR #190: Cannot upload bootloader in DFU mode

benoit-girard, 2019-07-12:
Using Ubuntu 16.04, and arduino IDE 1.8.9, I get the following message when trying to upload the bootloader while in DFU mode :

```
Arduino: 1.8.9 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

java.lang.NullPointerException
at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
at processing.app.Editor.lambda$handleBurnBootloader$65(Editor.java:2384)
at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.
```
the lsusb tells me that the card in DFU  mode is seen:

`girard@OldStar:~$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 003: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
`

OpusK, 2019-07-15:
Hi, @benoit-girard 

> java.lang.NullPointerException
at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
at processing.app.Editor.lambda$handleBurnBootloader$65(Editor.java:2384)
at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.

From the error log of Arduino, it seems that the dfu_util file updating the bootloader may not exist.

Please check if the file exists, or try reinstalling the OpenCR board package.

benoit-girard, 2019-07-15:
I fear I need help: where is this file supposed to be located?

OpusK, 2019-07-16:
It is located in `[The Arduino sketch folder you have set]/packages/OpenCR/tools`

closed 2019-07-17 by OpusK