ROBOTIS-GIT/OpenCR #193: Unexpected moving behaviour

Adel-kn, 2019-07-16:
Hello, 
i'm using Ubunto 16.04 to work on ROS kinetic version 1.
I am trying to apply a node to make the robot "turtlebot3" move "forward/backward" .
When I apply the node in the simulated environment "turtlebot3_fake" it works perfectly, but when i apply it on the robot, I got unexpected behaviour: e.g. when I give the command to robot to move forward only 1m then it keep moving  2.5m or 3m.
this problem happen also with rotational moving (rotating the robot around itself).
So I think the problem could be within the openCR itself.
Any help will be helpful.
thanks in advance

OpusK, 2019-07-17:
Hi, @Adel-kn 

I think there is a problem with the Turtlebot3 core example used in OpenCR and the node you created, rather than OpenCR itself.

Or it may be related to DYNAMIXEL encoders. In this regard, it would be better to issue an issue to the turtlebot3 repo site. 
Please inquire about the values set in the turtlebot3 core and the data sent from the node you created (presumably to be cmd_vel).

Adel-kn, 2019-07-17:
Thanks @OpusK for the reply.  
I charged the OpenCR with the Turtlebot3 core without any changing, but I didn't get your Idea about which problem could be between the turtlebot3 core and my node.

OpusK, 2019-07-18:
@Adel-kn 

Turtlebot3 If you look at the contents of core.ino file, you can see that cmd_vel is processed as below.

- [void commandVelocityCallback(const geometry_msgs::Twist& cmd_vel_msg)](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L168)

Please make sure that the commands you send in m units work well with this method.

I do not know much about the relationship between this core and the ros package, so @rjshim will answer in this regard.

rjshim, 2019-07-18:
@Adel-kn 

Please leave your TB3 related question at the following link. 
https://github.com/ROBOTIS-GIT/turtlebot3/issues/450

Thanks!
Ryan

closed 2019-08-19 by KyoungHwan2046