ROBOTIS-GIT/OpenCR #203: OpenCR Board no longer recognizing dynamixels after firmware update

Redslaya, 2019-09-03:
Remote PC OS: Ubuntu 16.04
OpenCR: 1.0
Dynamixel: XM430-W210-T x4
Raspberry Pi: Raspbian

This robot started as a turtlebot3 waffle pi  which i was converting into a tank with 4 motors. Initially I was able to get a third motor to be recognized by the openCR board with only the 4th motor not being seen. 

Eventually i figured out to use the USB _to_DXL firmware to get the dynamixel wizard 2.0 to see the servos. Once seen i was prompted to update the firmware on the 3 servos the board has been recognizing. After updating the dynamixel's firmware and restarting the board, now none of the motors can be found by the board either in the wizard or using the find_dynamixel firmware. 

I then updated the openCR library in the arduino IDE in an attempt to fix the problem to no avail.

Is there something I am missing to get the board to recognize the motors again? trying to use turtlebot_core no longer has the motors working, meaning there is definitley a conflict between the dynamixels and openCR.

KyoungHwan2046, 2019-09-04:
Hi, @Redslaya . 

Please check the ID of dynamixel each. And avoid using an identical ID for multiple dynamixels.

Redslaya, 2019-09-04:
@KyoungHwan2046 initially I had IDs 1-3. however now i cannot check or assign new IDs because the openCR board will not recognize any of the dynamixels

KyoungHwan2046, 2019-09-04:
@Redslaya  Do you mean that openCR doesn't recognize a dynamixel although only one dynamixel connect to openCR?

Redslaya, 2019-09-04:
@KyoungHwan2046 correct, even with only 1 dynamixel connected the board will not recognize that any dynamixel is connected through either  find_dynamixel or  2.0

KyoungHwan2046, 2019-09-05:
@Redslaya Have you ever change the scan option to check 'select all'? Please check this option with only 1 dynamixel connect to openCR.

Redslaya, 2019-09-05:
@KyoungHwan2046 yes I have it scanning all baudrates, all protocols.

KyoungHwan2046, 2019-09-06:
@Redslaya Would you please upload the 'usb_to_dxl' again? The wizard can looks like on scanning, although not 'usb_to_dxl' is uploaded but the others is uploaded.

Redslaya, 2019-09-09:
@KyoungHwan2046 After uploading the usb_to_dxl again, seeing the jump_fw, then scanning the dynamixels are still not showing up after a scan

KyoungHwan2046, 2019-09-10:
@Redslaya Is it possible to recovery operation with wizard 2.0?
You can check how to recovery the firmware of dynamixel on 
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery
.

KyoungHwan2046, 2019-09-10:
@Redslaya And, of course you checked the hardware state like cable connection, power, but check again please.

Redslaya, 2019-09-10:
@KyoungHwan2046 using a multi-meter on the outputs of the board through the cable to the motor i see 12v power from the middle pin, ground on the left most pin, and 5vish on the right. im thinking that maybe i need to run through the firmware-recovery, however when i try that step with the wizard it ask me to disconnect power and reapply power to the dynamixel, which i am unsure how to do without unplugging the whole thing. do i need to power it externally to the other pins on the dynamixel?

KyoungHwan2046, 2019-09-11:
@Redslaya You can power off to board using power switch without unplug the cable. Do you power to the board with external power supply or battery, right?  Let me know how to power the board and dynamixel of your system, please.

Redslaya, 2019-09-11:
@KyoungHwan2046 the board is connected directly to the power supply. the board is plugged into my linux machine via USB. the dynamixel is powered with a 3 pin cable from the TTL pins on the openCR board

KyoungHwan2046, 2019-09-16:
@Redslaya Please check the voltage of pins, when you turn off the switch of the openCR. It should be 0 volt.
So, If you try to run the recovery, connect the dynamixel to the openCR, turn on the power switch of the openCR(make sure that the power pin is 12V), run the wizard program, click the 'recovery' button, select your dynamixel model, select recovery port, turn off the power switch of the openCR and wait  5 or more seconds to make sure that the voltage of power pin is 0V, turn on the power switch of the openCR board and check that the recovery is running.

Redslaya, 2019-09-16:
@KyoungHwan2046 here are the steps i have done:

1. load usb_to_dxl firmware onto the openCR board
2. Ensured dynamixel XM430-210-T was corrected to a TTL port
3. verified with multimeter that 12v DC was connected
4. loaded Wizard 2.0 software and began recovery
5. Selected the XM-430-210 motor 
6. powered off the openCR board and ensured that 0V DC was was coming off pin 2. 3V off pin 3
7. powered on openCR board. 

recovery still does not recognize the dynamixel. beginning to think a RMA is necessary

KyoungHwan2046, 2019-09-20:
@Redslaya 
I hope the U2D2 will help you.

Redslaya, 2019-09-25:
From ROBOTIS:

It'll be certainly convenient if OpenCR or OpenCM9.04 can be a substitute for U2D2 or USB2Dynamixel in firmware management.

However, the firmware recovery or update via usb_to_dxl example can be unstable in some cases so it caused a dispute when the example was first uploaded.

Although the usb_to_dxl example was disclosed in order to help firmware management, I think it should be taken as a temporary measurement as it is not fully compatible with ROBOTIS software features and especially should not be used when testing DYNAMIXELs with high baudrate.

Our engineers are aware of this and I hope(as much as you do) it can be redesigned to support more stable firmware management.

This might happen earlier than we are expecting :)

closed 2019-09-25 by Redslaya

swiz23, 2021-03-02:
I realize this thread is quite old, but I've actually been playing around with getting the OpenCM9.04 board working as a substitute for the U2D2. So far, I've implemented a small sketch (similar to the usb_to_dxl sketch for the opencr) that can successfully be used with the Dynamixel Wizard 2.0 tool, and I was able to use it to recover dynamixel firmware. There were a couple catches though.

1) When using with the Dynamixel Wizard 2.0 tool, I'd recommend operating at a lower baudrate - 57600 bps worked great for me. 115200 worked well also with the occasional timeout. However, I got time outs quite often with 1 Mbps.

2) When doing firmware recovery, instead of using a usb cable going to the microusb port on the OpenCM9, I used an FTDI cable going from my computer's usb port to TX2, RX2, and ground (though I'm sure TX3 and RX3 would work just the same). I then altered my sketch to communicate over Serial2 as opposed to Serial. Then recovering firmware was a breeze (I tested multiple times on an XM430-W350 servo, but I imagine if it works on that servo, it should work for any dynamixel compatible with the Wizard).

Sketch can be found below.
[passthrough.zip](https://github.com/ROBOTIS-GIT/OpenCR/files/6071146/passthrough.zip)

ROBOTIS-Will, 2021-03-03:
Hi @swiz23 
Thank you for sharing the useful sketch.
In the latest OpenCM9.04 firmware update in R+Manager 2.0, bypass mode is supported so that DYNAMIXEL connected to OpenCM9.04 can be detected in the DYNAMIXEL Wizard 2.0.
I do not have much detail at this moment, but I'll get back to you with more information about this feature.
Thank you.

swiz23, 2021-03-03:
Oh! that's awesome! I'll have to try that out. Does the firmware also allow you to do firmware recovery?

ROBOTIS-Will, 2021-03-03:
Yes, firmware recovery is also supported :)

swiz23, 2021-03-03:
I see that R+ Manager 2.0 is not supported in Linux. Is the bypass firmware open source? If not, is there a binary somewhere that I can use linux to flash onto the OpenCM9?

ROBOTIS-Will, 2021-03-04:
You're right, there are some limitations in using OpenCM9.04 as a communication converter.
In fact, the OpenCM9.04 cannot be a complete substitute for U2D2 as each product designed in a specific way to serve their own purpose.
We're hoping that this firmware update could help some developers managing their DYNAMIXEL in more reasonable way.