ROBOTIS-GIT/OpenCR #209: ROS2 Firmware compilation error in Arduino IDE

giovanidiniz, 2019-10-07:
Hey guys,
I've been working onthe turtlebot3 burger ROS2 examples and I can't seem to figure out why while compiling the examples from the arduino IDE I get the following error.

==

Arduino: 1.8.9 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

sketch/roover.cpp: In function 'void RooverCore::begin(const char*)':
roover.cpp:311:63: error: invalid conversion from 'void (*)(uint16_t, uint8_t&) {aka void (*)(short unsigned int, unsigned char&)}' to 'DYNAMIXEL::userCallbackFunc {aka void (*)(short unsigned int, unsigned char&, void*)}' [-fpermissive]
   dxl_slave.setWriteCallbackFunc(dxl_slave_write_callback_func);
                                                               ^
In file included from /home/giovanidiniz/Arduino/libraries/Dynamixel2Arduino/src/Dynamixel2Arduino.h:22:0,
                 from sketch/roover_motor_driver.h:15,
                 from sketch/roover.h:16,
                 from sketch/roover.cpp:12:
/home/giovanidiniz/Arduino/libraries/Dynamixel2Arduino/src/utility/slave.h:60:10: note:   initializing argument 1 of 'void DYNAMIXEL::Slave::setWriteCallbackFunc(DYNAMIXEL::userCallbackFunc, void*)'
     void setWriteCallbackFunc(userCallbackFunc callback_func, void* callback_arg = nullptr);

          ^
exit status 1
invalid conversion from 'void (*)(uint16_t, uint8_t&) {aka void (*)(short unsigned int, unsigned char&)}' to 'DYNAMIXEL::userCallbackFunc {aka void (*)(short unsigned int, unsigned char&, void*)}' [-fpermissive]

==
roover.cpp is just the turtlebot3.cpp (I've been trying to teach my son a bit of programming so encouraging rewrite of the code). 

I have firmware written for almost all of turtlebot3's friends in ROS and want to port that to ROS2 but I can't get the examples to compile.

I am using the OpenCR board v 1.4.11 and v 0.3.0 of the Dynamixel2Arduino library (wherit seems to be the origin of the error).

Following the instructions on the e-manual, I get it to work nicely, but of course, so I think I just need to set the libraries to the right versions?

Thanks for the help!

OpusK, 2019-10-07:
Hello, @giovanidiniz 

I apologize for any inconvenience.
I have not released the turtlebot3 library for the Dynamixel2Arduino library API change.

OpenCR will be released soon. (To 1.4.12)
If you are not using OpenCR 1.4.12 or later, please use the Dynamixel2Arduino library below 0.3.0.

OpusK, 2019-10-07:
@giovanidiniz 

We have released OpenCR 1.4.12.
Please check the code of the released TB3 core.
You can see the change in #210.

giovanidiniz, 2019-10-08:
@OpusK

That worked! Thanks for the prompt help!

closed 2019-10-08 by giovanidiniz