ROBOTIS-GIT/OpenCR #211: What does this constant do?

edward17829991, 2019-10-08:
https://github.com/ROBOTIS-GIT/OpenCR/blob/92b7286aa1c73a765fc023aebb35840978d3b2d9/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_friends/turtlebot3_mecanum/turtlebot3_mecanum.ino#L166

As title, what is this "9.54"? Should I change it according to my setup?

Thanks in advance.

Edward Chen

KyoungHwan2046, 2019-10-11:
Hi, @edward17829991 .

Thank you for your questions.

The "9.54" is a value for making angular velocity unit to rpm unit.

The unit of "wheel_angular_velocity" is 'rad/sec, so multiplied by "9.54" to make it is rpm value.

KyoungHwan2046, 2019-10-21:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-10-21 by KyoungHwan2046