ROBOTIS-GIT/OpenCR #212: OpenCR is not connected to windows.

VanguardDream, 2019-10-11:
Hello, I am using OpenCR to making robot project.

at first few times it works okay but after some testing, openCR are not reconized by windows or i dont know about other OS.

so, i did some trouble shooting guides like DFU mode and so on.
but error is occur when i burn the bootloader on arduino IDE.

error message is 
"Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!"

and also, THE REAL PROBLEM that no com port (no connection) is not solve. 

is there any solution?

thank you.

KyoungHwan2046, 2019-10-16:
Hi @VanguardDream .

I'm sorry for the late reply.

This error message caused by various problems, so with only this message, I cannot figure out that what makes problem.

Would you write the full error message?

And I think this link https://github.com/ROBOTIS-GIT/OpenCR/issues/114#issuecomment-475366413 
could be help you.

VanguardDream, 2019-10-16:
Thank you for the information. but it dose not work too.

the entire error message for my case is below. (on Arduino IDE)
"
_Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 25604

Download	[                         ]   0%            0 bytes
Download	[=                        ]   7%         2048 bytes
Download	[===                      ]  15%         4096 bytes
Download	[=====                    ]  23%         6144 bytes
Download	[=======                  ]  31%         8192 bytes
Download	[=========                ]  39%        10240 bytes
Download	[===========              ]  47%        12288 bytes
Download	[=============            ]  55%        14336 bytes
Download	[===============          ]  63%        16384 bytes
Download	[=================        ]  71%        18432 bytes
Download	[===================      ]  79%        20480 bytes
Download	[=====================    ]  87%        22528 bytes
Download	[=======================  ]  95%        24576 bytes
Download	[=========================] 100%        25604 bytes
Download done.
File downloaded successfully
Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!_
"

i think my devie cant be wrote firmware exactly.

thank you have a good day.

KyoungHwan2046, 2019-10-16:
@VanguardDream 

> http://emanual.robotis.com/docs/en/parts/controller/opencr10/#driver-installoptional

With this manual, please update your dfu driver and burn the bootloader it again.

KyoungHwan2046, 2019-11-05:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-11-05 by KyoungHwan2046