ROBOTIS-GIT/OpenCR #214: LED pins are not as per e-manual

mkhuthir, 2019-10-15:
e-Manual stating that LED pins are:

[image]

Current firmware 1.2.3 is using:

Front_left = BDPIN_GPIO_6
Front_right = BDPIN_GPIO_8
Back_left   = BDPIN_GPIO_10
Back_right =BDPIN_GPIO_12

please check. Also I noticed in your video on how to connect LEDs to OpenCR, there are no resistors added 

[image]

I think we need to add a resistor for each LED. See below link for info.
http://www.resistorguide.com/resistor-for-led/

KyoungHwan2046, 2019-10-16:
Hi, @mkhuthir .

We just found out that the circuit of the video is wrong.

Your opinion is right. The resistor should be in the circuit.

We'll make correct the e-manual with uploading of correct circuit.

Thank you.

mkhuthir, 2019-10-16:
hi @KyoungHwan2046 

thank you very much; what about pin numbers ?

mkhuthir, 2019-10-16:
Thank you very much.

KyoungHwan2046, 2019-10-16:
@mkhuthir 

I have a question.

You said that
> current firmware 1.2.3 is using:
> Front_left = BDPIN_GPIO_6
> Front_right = BDPIN_GPIO_8
> Back_left = BDPIN_GPIO_10
> Back_right =BDPIN_GPIO_12
but I cannot found where this code is. 
Would you tell me where did you find this code?

And the last edit of led pin configuring is can find here.
> https://github.com/ROBOTIS-GIT/OpenCR/commit/190d9501aaba443e0e2380734daa07b2cc6a1081#diff-fe87ed179ef2b783508b98b1fb99ecdbR388

mkhuthir, 2019-10-16:
@KyoungHwan2046 

The pin numbers I mentioned are the actual active pins when I tested firmware 1.2.3 on my TurtleBot3 hardware. its not taken from code.

KyoungHwan2046, 2019-10-16:
@mkhuthir 

Could you please upload a picture that you connect pins?

Well, The openCR-binary code is based on the "openCR" code and you can see the configuration of led pins on 

>https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/turtlebot3/turtlebot3_sensor.cpp#L386
.

So, if the openCR with 'openCR-binary' code works different, than we will update on next release.

mkhuthir, 2019-10-16:
Im using the firmware binary from the following link:

https://github.com/ROBOTIS-GIT/OpenCR-Binaries/tree/master/turtlebot3/ROS1/latest

I will upload the photos soon.

mkhuthir, 2019-10-16:
Hello @KyoungHwan2046 

Sorry I could not share photos today because my TurtleBot3 not with me now. However I searched the code and I think I found the issue is in lines **62-79** of [chip.h](https://github.com/ROBOTIS-GIT/OpenCR/blob/113d56689d51fafc3b4547e5e77d2f3cf94b1621/arduino/opencr_arduino/opencr/variants/OpenCR/chip.h).

correct code should be:

```
#define BDPIN_GPIO_3            50
#define BDPIN_GPIO_4            51
#define BDPIN_GPIO_5            52
#define BDPIN_GPIO_6            53
#define BDPIN_GPIO_7            54
#define BDPIN_GPIO_8            55
#define BDPIN_GPIO_9            56
#define BDPIN_GPIO_10           57
#define BDPIN_GPIO_11           58
#define BDPIN_GPIO_12           59
#define BDPIN_GPIO_13           60
#define BDPIN_GPIO_14           61
#define BDPIN_GPIO_15           62
#define BDPIN_GPIO_16           63
#define BDPIN_GPIO_17           64
#define BDPIN_GPIO_18           65
#define BDPIN_GPIO_19           66
#define BDPIN_GPIO_20           67
```

[image]

mkhuthir, 2019-10-19:
@KyoungHwan2046 

Below are connection photos for your info. this is the correct connection after clearing confusion.
[image]
[image]
[image]

KyoungHwan2046, 2019-10-21:
Hi, @mkhuthir .

Thank you for your update. 

The pin map of the video is wrong and the modified video will be updated soon.

Once again, thank you for your help and supports.

mkhuthir, 2019-10-21:
Thank you very much @KyoungHwan2046

KyoungHwan2046, 2020-02-10:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2020-02-10 by KyoungHwan2046