ROBOTIS-GIT/OpenCR #217: Using OpenCR with USB Host Shield 2.0 to connect to PS4 controller

DesignToHelp, 2019-10-25:
Hi,

I am trying to use a PS4 controller with my dynamixel setup using the OpenCR 1.0 board. The USB Host Shield 2.0  library is used to talk to the controller, but the current library does not support Open CR 1.0, but it does support other ARM boards. Is it easy to modify the library to talk to Open CR? My guess is that it is just the SPI pins (and other pins) that need to be redefined, but any guidance regarding the same would be great.

https://github.com/felis/USB_Host_Shield_2.0

KyoungHwan2046, 2019-10-28:
Hi @DesignToHelp 
We only provide the solution of our product. 
So we don't provide the guidance of modify the USB Host Sheild 2.0 library.
You can check the SPI pins and SPI mode of openCR on 
>http://emanual.robotis.com/docs/en/parts/controller/opencr10/#arduino-connector

or at

>C:\Users\"username"\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.4.13\libraries\SPI

KyoungHwan2046, 2019-11-05:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-11-05 by KyoungHwan2046