ROBOTIS-GIT/OpenCR #218: how to get the odom with OPENCR + dynamixel motor ?

trandymage, 2019-11-03:
hi,everyone i am the new in ROS and using OPERCR board + 2 x dynamixel motor (XH540-W270)to bulid my robot. 
and now i have  some trouble.
i have followed the ROBOTIS e-Manual to setup my OPENCR board and have success to uploaded the turtlebot3_core.ino into the board.
i can also control the robot with  joy stick by the joy ROS package.
but when i try to use the SLAM gmapping (i'm sure the lidar is working, i have made a map by hector SLAM)i found that i can't get the twist liner odom, but the angular odom is working.
i check the source code 

* Calculate the odometry
*******************************************************************************/
bool calcOdometry(double diff_time)
{
  float* orientation;
  double wheel_l, wheel_r;      // rotation value of wheel [rad]
  double delta_s, theta, delta_theta;
  static double last_theta = 0.0;
  double v, w;                  // v = translational velocity [m/s], w = rotational velocity [rad/s]
  double step_time;

  wheel_l = wheel_r = 0.0;
  delta_s = delta_theta = theta = 0.0;
  v = w = 0.0;
  step_time = 0.0;

  step_time = diff_time;

  if (step_time == 0)
    return false;

  wheel_l = TICK2RAD * (double)last_diff_tick[LEFT];
  wheel_r = TICK2RAD * (double)last_diff_tick[RIGHT];

  if (isnan(wheel_l))
    wheel_l = 0.0;

  if (isnan(wheel_r))
    wheel_r = 0.0;

  delta_s     = WHEEL_RADIUS * (wheel_r + wheel_l) / 2.0;
  // theta = WHEEL_RADIUS * (wheel_r - wheel_l) / WHEEL_SEPARATION;  
  orientation = sensors.getOrientation();
  theta       = atan2f(orientation[1]*orientation[2] + orientation[0]*orientation[3], 
                0.5f - orientation[2]*orientation[2] - orientation[3]*orientation[3]);

  delta_theta = theta - last_theta;

  // compute odometric pose
  odom_pose[0] += delta_s * cos(odom_pose[2] + (delta_theta / 2.0));
  odom_pose[1] += delta_s * sin(odom_pose[2] + (delta_theta / 2.0));
  odom_pose[2] += delta_theta;

  // compute odometric instantaneouse velocity

  v = delta_s / step_time;
  w = delta_theta / step_time;

  odom_vel[0] = v;
  odom_vel[1] = 0.0;
  odom_vel[2] = w;

  last_velocity[LEFT]  = wheel_l / step_time;
  last_velocity[RIGHT] = wheel_r / step_time;
  last_theta = theta;

  return true;
}

and found that the angular odom can be get by the sensor,but the liner odom was only can be get by the encoder,  is that means the encoder in the motor didn't work? and how to check the status by the motor's encoder ? 
thank you

rjshim, 2019-11-04:
@trandymage 

Try below and let us know if you need more help.
$ rostopic echo /odom

Regards,
Ryan

trandymage, 2019-11-04:
@rjshim 

thanks for your kuick replyi run the code
but it only can see
---
header: 
  seq: 11845
  stamp: 
    secs: 1572860715
    nsecs: 202903015
  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: 0.000946186541114
      y: 0.0035163348075
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.168398588896
      w: 0.98571896553
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: -1.98682155315e-05
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 11846
  stamp: 
    secs: 1572860715
    nsecs: 235903015
  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: 0.000946186541114
      y: 0.0035163348075
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.168404370546
      w: 0.985718011856
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.00035582165583
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 11847
  stamp: 
    secs: 1572860715
    nsecs: 268903015
  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: 0.000946186541114
      y: 0.0035163348075
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.168396562338
      w: 0.985719323158
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: -0.000480449554743
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
i try to move and turn the robot but it only can see the angular odom form the sensor
if i let the robot leave the ground and make the wheel turn ,the angular odom also disappeared
so i guess the angular odom was got from the gryo sensor not from the encoder in the motor.

i think my issues is just like

https://github.com/ROBOTIS-GIT/turtlebot3/issues/194

but my oprncr board and turtle core vision is new i think
this is my terminal mseeage

pi@pi-CF-NX2RWJCS:~$ export TURTLEBOT3_MODEL=burger
pi@pi-CF-NX2RWJCS:~$ roslaunch turtlebot3_bringup turtlebot3_core.launch
... logging to /home/pi/.ros/log/acdd2dea-fee6-11e9-8310-e09d315f5a20/roslaunch-pi-CF-NX2RWJCS-3344.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pi-CF-NX2RWJCS:40905/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /turtlebot3_core/baud: 57600
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

auto-starting new master
process[master]: started with pid [3355]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to acdd2dea-fee6-11e9-8310-e09d315f5a20
process[rosout-1]: started with pid [3368]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [3371]
[INFO] [1572860232.589744]: ROS Serial Python Node
[INFO] [1572860232.598879]: Connecting to /dev/ttyACM0 at 57600 baud
[INFO] [1572860234.715468]: Note: publish buffer size is 1024 bytes
[INFO] [1572860234.715980]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1572860234.719061]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1572860234.751197]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1572860234.754816]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1572860234.766554]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1572860234.770275]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1572860234.773552]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1572860234.778363]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1572860235.384850]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1572860235.389974]: Note: subscribe buffer size is 1024 bytes
[INFO] [1572860235.390403]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1572860235.394825]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1572860235.402133]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1572860235.407541]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1572860237.834364]: Setup TF on Odometry [odom]
[INFO] [1572860237.836671]: Setup TF on IMU [imu_link]
[INFO] [1572860237.838813]: Setup TF on MagneticField [mag_link]
[INFO] [1572860237.840927]: Setup TF on JointState [base_link]
[INFO] [1572860237.855330]: --------------------------
[INFO] [1572860237.857888]: Connected to OpenCR board!
[INFO] [1572860237.860530]: This core(v1.2.3) is compatible with TB3 Burger
[INFO] [1572860237.862530]: --------------------------
[INFO] [1572860237.864419]: Start Calibration of Gyro
[INFO] [1572860240.396046]: Calibration End

trandymage, 2019-12-19:
@rjshim
sorry for the late report
i have do some test whit my robot
and i noticed that my robot's odom topic sometimes get freezed
just like
[image]

the odom linear freezed at this speed and i can't fix that untile re-upload turtlebot core.ino in to opencr.
i don't konw the reason why it happend.
please help me to fix that

KyoungHwan2046, 2020-01-28:
@trandymage 
I'm sorry for the late reply. 
Is the only odom linear still get freezed?

trandymage, 2020-01-28:
＠KyoungHwan2046
thanks for the reply.
yes,it still freezed when i start the turtlebot_core launch package,unless i reburn the turtlebot_core.ino into the OPENCR board.and it will freeze soon.
 i try to fix this problem by change another OPENCR board or change the XM540 motor but it didn't work.
I saw the  turtlebot_core.ino and found that the angular odom was picked up by the IMU sensor on the OPENCR board,and the linear odom was picked up by the motor encoder. so i think this is the reason why the angular odom didn't freezed but only the linear odom freezed.

please help me to fix that,thank you

KyoungHwan2046, 2020-01-28:
@trandymage 

I will check this problem as soon as possible.

rjshim, 2020-02-04:
@trandymage 

We tested our source code works without issues. Also, we do not support users' own customized robots. First build up a turtlebot as written on our instruction and customize it after if needed. 

Ryan.

KyoungHwan2046, 2020-02-10:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2020-02-10 by KyoungHwan2046