ROBOTIS-GIT/OpenCR #219: Failed to open Port Error (Fail to jump to boot)

DesignToHelp, 2019-11-04:
Any code i try to upload gives the following error. I tried different boards too.

hComm err : \
Fail to open port 1 : COM5
Fail to jump to boot

It says Done Uploading but I cannot open serial terminal

[image]

KyoungHwan2046, 2019-11-05:
Hi @DesignToHelp 

Please check below.

1. Is the openCR power on?
2. Is the openCR connected to pc?
3. Do you select the right comport on Arduino IDE?

You can get help on below page.
>https://www.arduino.cc/en/Guide/ArduinoUno

DesignToHelp, 2019-11-05:
The answer to all of those is YES. I was able to solve the issue by Holding down SW2 button, pressing RESET button, releasing RESET button and then releasing SW2 button. 

Then windows started recognizing the board. But, why did it happen in the first place?

KyoungHwan2046, 2019-11-07:
Hi @DesignToHelp 

I'm sorry for the late reply.

There was a problem with your openCR upload.

And I'm not sure what kind of problem was because the information of error is not enough.

What you did is recovery mode and you can check on below. 

>http://emanual.robotis.com/docs/en/parts/controller/opencr10/#references

KyoungHwan2046, 2019-11-20:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-11-20 by KyoungHwan2046