ROBOTIS-GIT/OpenCR #220: install opencr package via board manager

techjjun, 2019-11-08:
I'm using Nvidia XAVIER board as SBC and following emanual.robotic to install opencr package

in arduino IDE.  but it looks like unmet operating system and wanna know what it is..

[image]

KyoungHwan2046, 2019-11-12:
Hi @techjjun 

I'm sorry for the late reply.

We will checking the installation with board manager on the arm platform.

techjjun, 2019-11-15:
Thanks for the reply, I'll keep waiting for the respond.

KyoungHwan2046, 2019-11-20:
Hi @techjjun 

I'm sorry for the late reply.

We do not support the installation on the arm platform like XAVIER or raspberry pi. 

So, please use PC or laptop to use openCR.

closed 2019-12-15 by ROBOTIS-Will

robotiscustomer, 2020-04-06:
Hi, I am confused about the advice given above. We know that in the Turtlebot 3, raspberry pi is used with OpenCR. Why it was mentioned that it is not supported?  I recall seeing the idea that OpenCR only supports Intel PC somewhere else. Under what situation is it true?

ROBOTIS-Will, 2020-04-08:
@robotiscustomer 
The OpenCR package does not support ARM processor, which means you cannot build your source code using Arduino IDE on ARM based SBC such as RPi or Xavier. However, you can still download the compiled binary file using [OpenCR downloader](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#downloader).

robotiscustomer, 2020-04-08:
Thanks Will.