ROBOTIS-GIT/OpenCR #221: Bootloader not working  for OpenCR 1.0 borad

sandiprakhasiya, 2019-12-04:
I have  compiled the bootloader  on X86 machine with arm tool chain  gcc-arm-none-eabi (15:4.9.3+svn231177-1) .  After successful compilation i am able to flash bootloader on OpenCR1.0 board of turtlebot3 by following command .
# sudo dfu-util -d 0483:df11 -a 0 -s 0x08000000 -D ./opencr_boot.bin
 But It seems boot loader  is not working even status LED not blinking . 

Compilation logs and Flashing process logs is attached
[OpenCR1.0_Bootloader_Logs.zip](https://github.com/ROBOTIS-GIT/OpenCR/files/3920172/OpenCR1.0_Bootloader_Logs.zip)

ROBOTIS-Will, 2019-12-09:
@sandiprakhasiya 
Bootloader is already installed in the board and you do not have to burn it unless it is corrupted.
Please upload the TurtleBot3 sketch to OpenCR with Arduino in order to run your TurtleBot3 with OpenCR.
Thank you.

sandiprakhasiya, 2019-12-09:
> @sandiprakhasiya
> Bootloader is already installed in the board and you do not have to burn it unless it is corrupted.
> Please upload the TurtleBot3 sketch to OpenCR with Arduino in order to run your TurtleBot3 with OpenCR.
> Thank you.

 I want to modify the boot loader for our customer so we first we are trying to build from the source and burn on turtleboat3 .   Can you please help us to find out where am going wrong .

ROBOTIS-Will, 2019-12-12:
@sandiprakhasiya 
Customizing the bootloader is not recommended as it will void the warranty of the product.
Therefore, please understand that we cannot provide further help on this issue.
Thank you.

closed 2019-12-12 by ROBOTIS-Will