ROBOTIS-GIT/OpenCR #223: Debugging

vanem, 2019-12-24:
The OpenCR board has a JTAG connector. However I have found no info how to use it. How could I use ST link to debug for example?

KyoungHwan2046, 2019-12-27:
Hi @vanem 

We are not support the using of jtag with arduino IDE yet. 

You can use jtag with openCR code and another IDE, like a Truestudio, IAR, KEIL, or something else.

Thank you.

KyoungHwan2046, 2020-01-06:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2020-01-06 by KyoungHwan2046

Omar-Ismail, 2022-10-12:
Is it possible to know which type of connection could be made to the JTAG/SWD connector? The normal jumper wire that comes with any ST-Link V2 does not fit.

[image]

dtscbs, 2024-07-11:
This question was asked and answered several years ago.  Is JTAG support still not supported?