ROBOTIS-GIT/OpenCR #225: Adjusting Wheel Size on Turtlebot3 Burger

esch98, 2020-01-29:
I am new to ROS and OpenCR. I am working with a Turtlebot3 Burger and want to increase the size of the wheels for better ground clearance. I took a look at modifications that would need to be made to the OpneCR source so that the new wheels don't throw off the robot's pose estimation.

I found the WHEEL_RADIUS definition in turtlebot3_burger.h, which seems to be the correct setting to modify. However, looking through turtlebot3_core.ino, I also found a reference to the wheel circumference (listed as 0.207) that was simply entered as a number and did not reference the WHEEL_RADIUS (or any other) definition. These were the only two instances I was able to find that mentioned the wheel radius, but I was wondering if there are other settings that should be changed with a new wheel set.

KyoungHwan2046, 2020-01-30:
Hi, @esch98 

You should modify the "VELOCITY_CONSTANT_VALUE" and other parameters that 
related to velocity.

Thank you.

ROBOTIS-Will, 2020-01-31:
@esch98 
Odometry is calculated in OpenCR.
If you are going to use different size of wheels, it is recommended to update all parameters related to the wheel.
Those will be WHEEL_RADIUS, WHEEL_SEPARATION, TURNING_RADIUS, as well as [the Circumference of Wheel](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L575) value that you mentioned.
You can calculate the circumference of wheel by 2*Pi*WHEEL_RADIUS.

You might also want to update the URDF as there will be some changes in z axis.

KyoungHwan2046, 2020-02-10:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2020-02-10 by KyoungHwan2046

toukairinn9999, 2022-04-06:
@pitosalas 
That understanding is correct.
The steps required by OpenCR when changing wheel size or distance between wheels can be summarized as follows

1. change WHEEL_RADIUS, WHEEL_SEPARATION and TURNING_RADIUS in turtlebot3_burger.h

2. Change the circumference of the wheel in turtlebot3_core.ino. Here, the circumference can be calculated by 2Pi * WHEEl_RADIUS.
 _The default value of the circumference of burger is 0.207._

3. Install Arduino IDE on your PC and write the edited file (turtlebot3_core.ino). Please refer to the following e-Manual for this detail.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup
 _3.3. At the end of OpenCR Setup, there are instructions for using the Arduino IDE, click to expand and see the details._

I have changed these values and written them in the Arduino IDE and have confirmed that the speed of the wheel changes. Try this series of methods.

apusky, 2023-06-20:
> @pitosalas That understanding is correct. The steps required by OpenCR when changing wheel size or distance between wheels can be summarized as follows
>
> 1. change WHEEL_RADIUS, WHEEL_SEPARATION and [...]

Hello @toukairinn9999 ,
When you programmed the board with Arduino IDE, did you hear the beeping sounds?
Also, could you tell me what board version did you use?