ROBOTIS-GIT/OpenCR #227: Is Raspberry Pi 4 Model B fully compatible with Robotis products?

robotiscustomer, 2020-02-24:
Hi, I am new to OpenCR and Raspberry Pi. I am considering to use them to control dynamixels. Since Pi 4B is out, I am considering to buy it rather than the old Pi 3. Is the Pi 4B fully compatible with the OpenCR controller and other Robotis products without any modifications both hardware-wise and software-wise? If I put the Pi 4B in the TurtleBot3, will all the sample programs work out of the box without any modifications?

ROBOTIS-Will, 2020-02-28:
@robotiscustomer 
The new RPi4 can be used with TurtleBot3. 
However, there will be a new Raspbian image for RPi4. The image will soon be released via ROBOTIS eManual.
[http://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup](http://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup)
Thank you.

robotiscustomer, 2020-02-29:
@ROBOTIS-Will 
Thank you.

andreasBihlmaier, 2020-02-29:
Following the SBC Setup instructions (using the Ubuntu 18.04 image), everything works fine for me so far using the Raspberry Pi 4 instead of the Raspberry Pi 3+ shipped with the turtlebot3.

robotiscustomer, 2020-03-01:
@andreasBihlmaier 
Thank you. Which robot platform have you tested?

Actually, is it true that any SBC or controller that can run Ubuntu 18.04 will be able to connect to OpenCR to control the dynamixels without any modifications?

andreasBihlmaier, 2020-03-01:
I tested on Waffle Pi with OpenManipulator and ROS1.

robotiscustomer, 2020-03-01:
@andreasBihlmaier 
Thank you. I guess it might work on the Burger.

ROBOTIS-Will, 2020-03-17:
@robotiscustomer @andreasBihlmaier 
If you want to try the TurtleBot3 RPi4 ROS1 Raspbian image, please try [this Raspbian image](https://drive.google.com/file/d/1MzSjDoDcCGV3hTet6iVL9tKrIRn7nzIT/view?usp=drive_open).
The contents in this image may change without prior notice, but please feel free to give us some feedback.
Thank you for your patience and support.

andreasBihlmaier, 2020-03-17:
Thank you for the support. Just to let everyone working in the same direction know:
On ROS**1** (melodic) the Turtlebot3 + OpenManipulator works fine using the 64-bit Ubuntu 18.04 (aarch64) images. I will try to write up on the weekend, where I had to deviate from the current documentation and which other (hard to figure out, but minor in the end) changes were required.

ROBOTIS-Will, 2020-03-23:
@andreasBihlmaier If you can share the documentation, I'll hand it to the ROBOTIS eManual admin @ROBOTIS-David to update the content. Thank you!

closed 2020-05-08 by ROBOTIS-Will

amaheshwari2509, 2021-05-20:
I am using OpenCR for Turtlebot3 on raspberry pi4 for ROS2. I am using Turtlebot3 ROS2 with OpenCR library from ROS(1). But It is not allowing torque enable to the motor. Is it because OpenCR library of ROS(1) is not compatible with Turtlebot3 ROS2 library ? 

I am bit reluctant to switch to OpenCR ROS 2 package because I have earlier customized my Turtlebot3 with 3 motors (1 front and 2 rear) instead of default 2 motors.

ROBOTIS-Will, 2021-05-24:
@amaheshwari2509 
The OpenCR ROS1 firmware is not compatible with ROS2 as there are several major changes.
You'll need to modify the TurtleBot3 ROS2 firmware from Arduino IDE in order to operate your customized system.
Thank you.

amaheshwari2509, 2021-05-25:
Hello, Yes I did the same by changing motor driver Cpp amd h file as we add additional dynamixal motor in Turtlebot3 ROS2 OpenCR firmware. But it's not working. I have explained the issue over here : https://github.com/ROBOTIS-GIT/turtlebot3/issues/744. Can you please assist me ?

Thanks

ROBOTIS-Will, 2021-05-26:
Hi @amaheshwari2509 
I'm afraid that we do not provide assistance for product modification, but I'll provide a general explanation about how OpenCR work with ROS2 in turtlebot3 issue thread #744(https://github.com/ROBOTIS-GIT/turtlebot3/issues/744)

amaheshwari2509, 2021-05-26:
Thanks @ROBOTIS-Will. I understand. I believe general explanation will be suffice so that I can customize it accordingly by myself . Looking forward to read you comment in https://github.com/ROBOTIS-GIT/turtlebot3/issues/744

ROBOTIS-Will, 2021-05-26:
Thank you for your understanding and let us know if you need additional help for the customization.
FYI, you can also share your project on [community.robotis.us](https://community.robotis.us/) and get feedback/help from other developers.
Thank you.