ROBOTIS-GIT/OpenCR #228: turtlebot3_with_open_manipulator_core missing for ROS2

andreasBihlmaier, 2020-02-29:
There seems to be no equivalent for the turtlebot3+openmanipulator ROS1 code (https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_with_open_manipulator/turtlebot3_with_open_manipulator_core), at least I could not find it (e.g. in https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples).

Likely the following issue is related to this problem: https://github.com/ROBOTIS-GIT/OpenCR-Binaries/issues/7

timodd, 2020-03-11:
Hi,
I'm in the same situation as described in the related issue.
I'd like to use ROS2 with waffle pi + om and I would very much appreciate a solution for this.

ROBOTIS-Will, 2020-03-13:
Hi, the manipulation firmware example for TurteBot3 waffle in ROS 2 will be available after Foxy is released in this May.
Thank you.

andreasBihlmaier, 2020-03-13:
Thank you for confirming that this is not available at the moment and providing a timeline for when support will be added.

ROBOTIS-Will, 2020-03-17:
@andreasBihlmaier 
We're currently revising manipulation packages. ROS 2 contents will be reviewed along with Foxy support, so I'd expect it within a couple month.
Thank you.

chiuau, 2021-03-19:
> Hi, the manipulation firmware example for TurteBot3 waffle in ROS 2 will be available after Foxy is released in this May.
> Thank you.

Could you please tell me where I can download the firmware for the OpenCR's firmware for TurteBot3 waffle with open manipulator in ROS 2?  Thanks.

ROBOTIS-Will, 2021-03-23:
@chiuau 
Thank you for the inquiry and apologize for the delay in this issue.
This firmware is scheduled to be developed with [Home Service Challenge](https://emanual.robotis.com/docs/en/platform/turtlebot3/home_service_challenge/) update in this 2Q.
Foxy will be the first priority in ROS2.

joaoramosipl, 2021-08-06:
Hello. Can you please help me with ROS2 Foxy? I can't still find any package for this version, to be able to use OpenManipulator with the TurtleBot3 Waffle.
Thank you.

ROBOTIS-Will, 2021-08-09:
Hi @joaoramosipl 
Thank you for your interest.
I've been working on the latest OpenMANIPULATOR-X documentation, and TB3 Manipulation Noetic update will be followed by.
There are some delays in ROS2 support and needs some time to resolve them, for example: https://github.com/ROBOTIS-GIT/turtlebot3/issues/744
I'm expecting ROS2 Foxy TB3 manipulation done in September with some important updates.
Thank you.

eslamsalahelsheikh, 2021-12-17:
Hello, is there any updates on this?

ROBOTIS-Will, 2021-12-20:
Hi @eslamsalahelsheikh 
I'm sorry about the delay in TB3 Manipulation Foxy update.
There are some imminent updates required and had to postpone the TB3 manipulation update to the first quarter of 2022.
Sorry about the inconvenience, but we'll try to update as early as possible.
Thanks!

Torsten12, 2022-02-28:
Hey @ROBOTIS-Will 
is there any possibility that you can give a status of the current state? 
We would like to use that hardware in a current project and are a little bit on a break because of this issue. 

But thanks for your work! Really appreciate it.

shiba-8ro, 2022-06-02:
ROS1 Noetic support will be completed by ROBOTIS Japan team in June. 
But each person want to use ROS2 Foxy. Will Son, let's hurry up!

ROBOTIS-Will, 2022-06-02:
I'm sorry about the delayed reply.
Basic packages(SLAM, Navigation, Teleop, Simulation) for Humble will be updated from tomorrow.
TB3 Manipulation for Foxy is scheduled by 7/14, but it might be released a bit earlier.
At the moment, scheduled update is as below

|Packages|Foxy|Humble|
|:------:|:------:|:----:|
|Autorace|-|-|
|Machine Learning|-|-|
|Manipulation|||
| ㄴ Simulation|6/16|7/28(Ignition)|
| ㄴ SLAM|7/14|7/28|
| ㄴ Navigation|7/14|7/28|
| ㄴ H.S.C|-|-|

Hopefully I can release TB3 Manipulation for Foxy by this month as well as Humble.
Sorry about the delay.

Vitus012, 2022-06-16:
How far are you?
I want to use this hardware, specifically in foxy, for a student project.

ROBOTIS-Will, 2022-06-17:
Hi @Vitus012 

For now, Foxy simulation on Gazebo is almost done and going to apply ros2_control for controlling the manipulator.
Later I might heavily reduce the details of the OpenMANIPULATOR-X and remove the dependency for the open_manipulator packages as there are many developers who didn't like the dependencies to other packages.
Thanks!

[image]

andreasBihlmaier, 2022-07-06:
@ROBOTIS-Will great to see the progress. I just updated my last machine to jammy and thus no more ROS1, only ROS2 (humble) from here on forward. Put differently, I'm now unable to use TB3 and OpenManipulator-X unless ROS2 support is in place. Likely others are in a similar situation with the release of jammy and humble respectively.

david-peitz, 2022-09-07:
@ROBOTIS-Will Is it possible to get an update on the release of the TB3 Manipulation for Foxy?

ROBOTIS-Will, 2022-09-08:
Hi, 
I'm sorry about the delayed updates.
The current status for TB3 Manipulation update is in progress in following tasks and will be completed by September 30th.

- OpenCR Firmware update (WIP, #316)
- Moveit environment setup (WIP)
- [TurtleBot3 Manipulation package](https://github.com/ROBOTIS-GIT/turtlebot3_manipulation/tree/foxy-devel) implementation ([WIP](url))
- Gazebo environment setup (WIP)
- eManual update (by September 30th)

RubenCarreira99, 2022-10-01:
Hi, any updates on this matter?

david-peitz, 2022-10-02:
@ROBOTIS-Will We all wait with bated breath....

tielef, 2022-10-24:
Hello, any updates on this issue? I want to use the TurtleBot3 with OpenManipulator in Foxy for a university project.

yun-goon, 2025-03-11:
Sorry for the late response. This issue was created a long time ago.

Please try again with the latest version, and if you still encounter any problems, feel free to open a new issue.

Thank you for your understanding!

closed 2025-03-11 by yun-goon