ROBOTIS-GIT/OpenCR #229: Turtlebot3 ROS2 OpenCR in Arduino IDE

giovanidiniz, 2020-03-01:
@OpusK,

I've been trying to compile the turtlebot3 firmware in the Arduino IDE but I think there is something I am missing in the process. Whenever I compile the burger firmware in the IDE and launch the bringup script in ROS2 I get the following error message:

[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
**[turtlebot3_ros-3] [ERROR] [turtlebot3_node]: Failed connection with Devices**

and the node initialization fails... If I use the binary provided in the e-manual I can connect with the node without a problem!!,

I'm guessing there is an issue with the firmware on github? Maybe I need to use a specific board version? Dynamixel2Arduno library version?

Any help would be much appreciated

Thanks!

OpusK, 2020-03-02:
Hi, @giovanidiniz 
We apologize for the inconvenience.

If you used the latest version of Dynamixel2Arduino, you may have compile errors.
Since TB3 F/W has not been modified for the new version of Dynamixel2Arduino, please use version 0.3.0 of Dynamixel2Arduino.

giovanidiniz, 2020-03-05:
No worries at all! I figured it was something I had missed on readme.

That worked, thanks!

closed 2020-03-05 by giovanidiniz

yzprofile, 2020-04-23:
@giovanidiniz hi, what did you missed ? 

i'm  facing the same error .

thx

yzprofile, 2020-04-24:
> @giovanidiniz hi, what did you missed ?
> 
> i'm facing the same error .
> 
> thx

found out why... 
only ver: 1.4.10 works well.

giovanidiniz, 2020-04-24:
Yes, you are right.

The LTS version of the SDK wasn't supported when I last tried to work on
that.