ROBOTIS-GIT/OpenCR #230: DXL rotation orientation

giovanidiniz, 2020-03-05:
@OpusK 
Thanks for the feedback on the v of Dynamixel2Arduino lib. I managed to compile the firmware on the OpenCR board and test it.

During these test however (running with both the waffle and burger FW) the orientation of the motors seems to be messed up. 

I understand the positive rotation of the servos is with the axis points outward of the horn. I think there is either a configuration issue with my DXL firmware versions or somewhere in the OpenCR firmware. 

Any ideas?

OpusK, 2020-03-06:
@giovanidiniz 

OpenCR firmware does not set the motor direction.
It simply calculates the value passed from the ROS2 package as the speed value to put into the motor.
Compare your current motor ID(LEFT/RIGHT) with the e-manual.

@ROBOTIS-Will, @rjshim, any ideas?

giovanidiniz, 2020-03-06:
The curious thing is when I run the burger FW on XL430-250's the motors spin exactly the opposite direction as expected. 
While running the waffle FW on XM430-210's I get completely wrong rotation orientation. 

You mean the DXL_MOTOR_ID_RIGHT under the motor_driver.cpp file? I set those accordingly

OpusK, 2020-03-06:
> You mean the DXL_MOTOR_ID_RIGHT under the motor_driver.cpp file? I set those accordingly

Yes, that's right!

> While running the waffle FW on XM430-210's I get completely wrong rotation orientation.

That's weird.
As you can see by searching for the code with the word waffle, the TB3 ROS2 firmware only handles buger and waffle for parameters that are not oriented.

The package will check with others. But finally, can you check the direction of OpenCR and the direction of the motor? And, if you have a photo, it will help them more (including DXL ID).
- [TB3 components](http://emanual.robotis.com/docs/en/platform/turtlebot3/specifications/#components)

giovanidiniz, 2020-03-06:
I setup the hardware exactly the same orientation as described in the e-manual. 
I wonder if there is a specific DXL configuration required, e.g., firmware version, operating mode...

I tested out the motors using the example i_velocity OpenCR sketch in the Arduino IDE. When I run the on my XL430's both motors start spinning forward (my right motor in negative direction -which is what the code says and the left one in positive direction-which is odd...
The XM430's on the other hand, both spin in the negative rotation axis direction (right rotates the wheel forward and the left, backwards)

That test makes me think that I'm missing some configuration on the DXL setup.

Right now, all I did with the motors was set mode to velocity, set the ID and change BPS. All done with the Arduino sketches offered as example in the IDE

ROBOTIS-Will, 2020-03-06:
@giovanidiniz 
By default, positive rotational direction of DYNAMIXEL is CCW as you already understand.
In case of TurtleBot3, the rotational direction for the right wheel is converted to the other direction.
This is configured from [turtlebot3_setup_motor Arduino example](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3), and you can check that Drive Mode bit 0 is set to 1 for the right wheel in below code.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_setup/turtlebot3_setup_motor/turtlebot3_setup_motor.ino#L340

giovanidiniz, 2020-03-06:
Ha! That was it. Had completely forgotten about that step. Man, you guys rock. 
Thank you so much!!

closed 2020-03-06 by giovanidiniz