ROBOTIS-GIT/OpenCR #236: OpenCR needing to upload twice to work

robotiscustomer, 2020-04-28:
Hi,

When following the e-Manual of the OpenManipulator-X, I noticed that almost all the time I need to perform upload twice to OpenCR in order to succeed the uploading process. In the first trial, the error is always:

ser_open: unable to open port: No such file or directory
opencr_ld ver 1.0.2
opencr_ld_main
>>
file name : /tmp/arduino_build_802820/usb_to_dxl.ino.bin
file size : 111 KB
Fail to open port 1

Without doing anything, when I pressed the upload button the second time, then upload succeed. How come? Is there something wrong with my OpenCR board?

robotiscustomer, 2020-05-14:
Hi, where is this thread transferred to? Link please.

ROBOTIS-Will, 2020-05-15:
You are already at the transferred thread that I moved from OpenMANIPULATOR to OpenCR.
In order to make sure that OpenCR is ready for the new sketch download, please enter the boot mode before uploading a sketch from Arduino IDE.
Please try below steps to download and let me know if this doesn't work.
You can follow below steps to enter the boot mode.

1. Press and hold SW2 button next to the micro USB port.
2. Press the Reset button
3. Release the SW2 button
4. Check if Status LED is blinking
5. Proceed uploading Arduino example 

Thank you.

robotiscustomer, 2020-05-21:
I was doing it under Linux. I bought the OpenCR as part of the TurtleBot3 kit. Does the standalone OpenCR board has the issue I and EndlessLoops mentioned?

ROBOTIS-Will, 2020-05-21:
There were a few cases in some systems where OpenCR needs to be put in boot mode before uploading the code, but the exact reason for it is not known yet.

robotiscustomer, 2020-05-22:
Thanks Will.

closed 2020-05-22 by ROBOTIS-Will