ROBOTIS-GIT/OpenCR #237: Can upload on Windows but not on Mac version of Arduino IDE

dmeglan, 2020-05-06:
I recently got an OpenCR board and a number of Dynamixels. My preferred dev machine is a mac, but I get a failed upload onto the board with it (programs compile). On the PC I can compile/upload and run examples like p_Monitor. After I have uploaded that, I can reconnect to the mac and interact with the newly loaded program using the Serial Monitor in the Arduino IDE. So the mac serial port connection to the OpenCR works, just not being able to upload the compiled program.

The error message I am getting is below. I get that here is a java error in the port opening but I don't know how to debug this. The port is definitely there -- you can see it from the command line. And it works since I can interact with p_Monitor. It's the program uploading that is failing.

I am wondering if there is some type of permissions issue. This is the newest MacBook Pro with the most recent OS and updates.

Guidance appreciated.

--dwight

Arduino: 1.8.12 (Mac OS X), Board: "OpenCR Board, OpenCR Bootloader"

Sketch uses 160976 bytes (20%) of program storage space. Maximum is 786432 bytes.
Global variables use 41452 bytes of dynamic memory.
processing.app.SerialException: Error opening serial port '/dev/cu.usbmodemFFFFFFFEFFFF1'.
	at processing.app.Serial.<init>(Serial.java:147)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$2.<init>(SerialMonitor.java:130)
	at processing.app.SerialMonitor.open(SerialMonitor.java:130)
	at processing.app.AbstractMonitor.resume(AbstractMonitor.java:132)
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2107)
	at processing.app.Editor.access$1300(Editor.java:117)
	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - /dev/cu.usbmodemFFFFFFFEFFFF1; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.<init>(Serial.java:136)
	... 8 more
Error opening serial port '/dev/cu.usbmodemFFFFFFFEFFFF1'.
opencr_ld ver 1.0.2
ser_open: unable to open port: No such file or directory
opencr_ld_main 
>>
file name : /var/folders/wm/l82y0ds95msdxlr_8w57jxhm0000gs/T/arduino_build_66828/p_Monitor.ino.bin 
file size : 158 KB
Fail to open port 1

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dmeglan, 2020-05-06:
I noticed the comment about setting things to Verbose in the Preferences so that I did that and expanded info it below. I did both expanded compile and upload errors even though I think only upload is the problem.
--dwight

Arduino: 1.8.12 (Mac OS X), Board: "OpenCR Board, OpenCR Bootloader"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dmeglan/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dmeglan/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dmeglan/Documents/Arduino/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -vid-pid=0483_5740 -ide-version=10812 -build-path /var/folders/wm/l82y0ds95msdxlr_8w57jxhm0000gs/T/arduino_build_66828 -warnings=none -build-cache /var/folders/wm/l82y0ds95msdxlr_8w57jxhm0000gs/T/arduino_cache_809245 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_tools.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_tools-1.0.0.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_gcc.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_gcc-5.4.0-2016q2.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -verbose /Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/libraries/OpenCR/examples/08. DynamixelWorkbench/p_Monitor/p_Monitor.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dmeglan/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dmeglan/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dmeglan/Documents/Arduino/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -vid-pid=0483_5740 -ide-version=10812 -build-path /var/folders/wm/l82y0ds95msdxlr_8w57jxhm0000gs/T/arduino_build_66828 -warnings=none -build-cache /var/folders/wm/l82y0ds95msdxlr_8w57jxhm0000gs/T/arduino_cache_809245 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_tools.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_tools-1.0.0.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_gcc.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_gcc-5.4.0-2016q2.path=/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -verbose /Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/libraries/OpenCR/examples/08. DynamixelWorkbench/p_Monitor/p_Monitor.ino
Using board 'OpenCR' from platform in folder: /Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15
Using core 'arduino' from platform in folder: /Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15
Detecting libraries used...
/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10812 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR/bsp/opencr -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR/bsp/opencr/include -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR/hw -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR/hw/driver -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR/hw/usb_cdc -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR/ -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/cores/arduino -I/Users/dmeglan/Library/Arduino15/packages/OpenCR/hardware/OpenCR/1.4.15/variants/OpenCR /var/folders/wm/l82y0ds95msdxlr_8w57jxhm0000gs/T/arduino_build_66828/sketch/p_Monitor.ino.cpp -o /dev/null
Alternatives for DynamixelWorkbench.h: [DynamixelWorkbench@0.1.0]
ResolveLibrary(DynamixelWorkbench.h)
  -> candidates: [DynamixelWorkbench@0.1.0]
/Users/dmeglan/Library/Arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10812 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/Users/dmeglan/Libra [... cut ...]
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2107)
	at processing.app.Editor.access$1300(Editor.java:117)
	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - /dev/cu.usbmodemFFFFFFFEFFFF1; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.<init>(Serial.java:136)
	... 8 more
Error opening serial port '/dev/cu.usbmodemFFFFFFFEFFFF1'.

dmeglan, 2020-05-06:
Another data point. I can compile/upload programs for a Sparkfun Redboard Turbo on this mac without a problem. So I am expecting this to be an issue specific to the OpenCR rather than a global serial port permissions problem.

ROBOTIS-Will, 2020-05-15:
Hi, is your issue something similar to #236?
Please try uploading the sketch while OpenCR is in boot mode and let us know.
Thank you.

dmeglan, 2020-05-16:
Doing the SW2 and reboot button press combination to put it that mode works. The sketch uploads and runs. Thanks.

ROBOTIS-Will, 2020-05-19:
@dmeglan Thank you for the feedback. Please let us know if there's any other issues to be resolved.
Hope you and your family are safe from the Covid-19.

closed 2020-05-19 by ROBOTIS-Will