ROBOTIS-GIT/OpenCR #239: Flashing issues

zentala, 2020-05-24:
**TL;DR**: after a few flashings it seems that flashing is working (diodes blinking, buzzer plays sound from examples that I flashed), but communication via serial port stopped. Is it any know firmware issue or device is broken? 

I bought this OpenCR with TB3 Waffle Pi. I was able to flash OpenCR a few times (around 5-10) via Arduino IDE, then I wasn't able to connect with ROS or flash it again.

**Example error 1:** 
I flashed example with music from buzzer (example 02.a) and I got following error (but music played):
```
Sketch uses 56444 bytes (7%) of program storage space. Maximum is 786432 bytes.
Global variables use 40028 bytes of dynamic memory.
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_283453/a_toneMelody.ino.bin 
file size : 55 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 1.005000 sec
flash_write : -1 : 1.886000 sec 
cmd_flash_fw_write_begin ERR : 0xF020
[FAIL] Download
```

**Example error 2:** 
LED example is also working but there is also flashing error (flashing via recovery mode):
```
Sketch uses 56460 bytes (7%) of program storage space. Maximum is 786432 bytes.
Global variables use 39968 bytes of dynamic memory.
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_311098/b_Blink_LED.ino.bin 
file size : 55 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 1.002000 sec
flash_write : 0 : 0.322000 sec 
CRC OK 554B6A 554B6A 0.002000 sec
[OK] Download 
jump_to_fw 
processing.app.SerialException: Error opening serial port '/dev/ttyACM0'.
	at processing.app.Serial.<init>(Serial.java:147)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$2.<init>(SerialMonitor.java:130)
	at processing.app.SerialMonitor.open(SerialMonitor.java:130)
	at processing.app.AbstractMonitor.resume(AbstractMonitor.java:132)
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2107)
	at processing.app.Editor.access$1300(Editor.java:117)
	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - /dev/ttyACM0; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.<init>(Serial.java:136)
	... 8 more
Error opening serial port '/dev/ttyACM0'.
```

**Example error 3:** 
When I am flashing original code for TB3 Waffle via Arduino IDE in recovery mode I got this error:
```
Sketch uses 184744 bytes (23%) of program storage space. Maximum is 786432 bytes.
Global variables use 46572 bytes of dynamic memory.
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_753415/turtlebot3_core.ino.bin 
file size : 181 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x00000000
Board Rev  : 0x00000000
>>
flash_erase : 0 : 1.003000 sec
flash_write : 0 : 1.683000 sec 
CRC OK 12957FC 12957FC 0.006000 sec
[OK] Download 
jump_to_fw 
processing.app.SerialException: Error opening serial port '/dev/ttyACM0'.
	at processing.app.Serial.<init>(Serial.java:147)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$2.<init>(SerialMonitor.java:130)
	at processing.app.SerialMonitor.open(SerialMonitor.java:130)
	at processing.app.AbstractMonitor.resume(AbstractMonitor.java:132)
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2107)
	at processing.app.Editor.access$1300(Editor.java:117)
	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - /dev/ttyACM0; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.<init>(Serial.java:136)
	... 8 more
Error opening serial port '/dev/ttyACM0'.
```
**Example error 4:** 
When I am flashing all the above examples without recovery mode Arduino IDE stuck in:
```
Sketch uses 56460 bytes (7%) of program storage space. Maximum is 786432 bytes.
Global variables use 39968 bytes of dynamic memory.
```
And then don't respond for a long time. I see message in Arduino `Uploading...` but nothing is happening. 
Diode `STATUS` is blinking all the time.

**Example error 5:** 
I have been trying to reintroduce original firmware via console too:
```
$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=waffle
$ rm -rf ./opencr_update.tar.bz2
$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 && tar -xvf opencr_update.tar.bz2 && cd ./opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr && cd ..
--2020-05-24 04:04:23--  https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2
Resolving github.com (github.com)... 140.82.118.3
Connecting to github.com (github.com)|140.82.118.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR-Binaries/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 [following]
--2020-05-24 04:04:24--  https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR-Binaries/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.192.133, 151.101.0.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.192.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 462386 (452K) [application/octet-stream]
Saving to: ‘opencr_update.tar.bz2.6’

opencr_update.tar.bz2.6                    100%[=======================================================================================>] 451.55K  1.48MB/s    in 0.3s    

2020-05-24 04:04:24 (1.48 MB/s) - ‘opencr_update.tar.bz2.6’ saved [462386/462386]

opencr_update/burger_turtlebot3_core.ino.bin
opencr_update/om_with_tb3.opencr
opencr_update/waffle_turtlebot3_core.ino.bin
opencr_update/opencr_ld_shell_arm
opencr_update/update.sh
opencr_update/
opencr_update/released_1.2.5.txt
opencr_update/waffle.opencr
opencr_update/opencr_ld_shell_x86
opencr_update/burger.opencr
opencr_update/open_manipulator_turtlebot3_core.ino.bin
x86_64
amd64
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main 
[  ] file name   	: waffle.opencr 
[  ] file size   	: 178 KB
[  ] fw_name     	: waffle 
[  ] fw_ver      	: 1.2.3 
ser_open: unable to open port: Device or resource busy
Fail to open port 1 : /dev/ttyACM0
[NG] Fail to jump to boot
[  ] file name   	: waffle.opencr 
[  ] file size   	: 178 KB
[  ] fw_name     	: waffle 
[  ] fw_ver      	: 1.2.3 
ser_open: unable to open port: Device or resource busy
Fail to open port 1 : /dev/ttyACM0
[NG] Fail to jump to boot
[  ] file name   	: waffle.opencr 
[  ] file size   	: 178 KB
[  ] fw_name     	: waffle 
[  ] fw_ver      	: 1.2.3 
ser_open: unable to open port: Device or resource busy
Fail to open port 1 : /dev/ttyACM0
[NG] Fail to jump to boot
```

**Example errror 6:** 
```roslaunch turtlebot3_bringup turtlebot3_core.launch                         ✔  04:04:25  
... logging to /home/zentala/.ros/log/a84d9ed0-9b4f-11ea-b97e-e159d4ff4551/roslaunch-zentala-XPS-8930-29004.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/zentala/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.0.120:43767/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

ROS_MASTER_URI=http://192.168.0.120:11311

process[turtlebot3_core-1]: started with pid [29023]
[INFO] [1590287608.072356, 0.000000]: ROS Serial Python Node
[INFO] [1590287608.082992, 0.000000]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1590287608.087832, 0.000000]: Error opening serial: [Errno 16] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'
```

The device is:
* powered on (via both: TB3 charger and USB), 
* connected to the computer (visible in the `/dev/ttyACM*`),
* `POWER` and `USB` diodes are always green

I tested on under two computers with Linux. 
Before device worked well with both of them.

ROBOTIS-Will, 2020-05-25:
Hi,
Thank you for sharing the various error logs. Since your OpenCR keeps failing to be connected to the port, please check if microUSB connector is damaged or detached from the circuit board.
Also try with different USB cables and see if it helps to resolve the issue.
Lastly, try [Bootloader Update](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#update-bootloader) on your PC and proceed until [Burn Bootloader](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#burn-bootloader) section.

zentala, 2020-05-27:
I tested with a few cables and I am sure they are not damaged. 

Burning a bootloader helped partially. After burning a bootloader and flashing a few additional times the same issue occurs. Sometimes I am event not able to flash bootloader again. But if I will wait 20-30minutes, reflash bootloader and then firmware again then the error doesn't occur anymore. But if I will reflash a few additional times in series error repeats and I need to wait again to reflash - firstly bootloader, then code.  So for me, this is a matter of time between flashing.

I suspect that my OpenCR is damaged or all OpenCRs had a low-quality flash memory, which is corrupting data when you flashing a few times in series.

ROBOTIS-Will, 2020-05-29:
Hi,
Still not sure what the cause of this issue is, and I don't get it when the board works again after 20~30minutes. If the firmware or bootloader is corrupted, waiting for some time would not fix the issue.
Have you tried rebooting your PC after the issue occurred?
If possible, please give us more information about your PC (CPU, OS version) you have tested.

zentala, 2020-06-05:
I tried to flash it it using 2 totally different machines (one of them is Dell XPS-8930). It is not a reason of this issue. 

Steps to reproduce:
1) Reflash OpenCR 10x in a row (eg every minute, during 10 minutes). It will crash showing above errors. 
2) Try to reflash the bootloader. This will fail. Try to reflash bootloader 10x in a row. It's failing.
3) Wait 30 minutes. 
4) Reflash bootloader again. Success! 
5) Reflash firmware. Succes. 
6) Reproduce 1st step. It will fail around 2nd-5th flashing.

ROBOTIS-Will, 2020-08-27:
Sorry, I'm still not able to reproduce this symptom.
Though I sometimes get below error, retrying sketch upload just works fine to me.

```
Sketch uses 56460 bytes (7%) of program storage space. Maximum is 786432 bytes.
Global variables use 39968 bytes of dynamic memory.
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_375876/b_Blink_LED.ino.bin 
file size : 55 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
cmd_read_board_name fail : 0xF020
An error occurred while uploading the sketch
```

closed 2020-11-02 by ROBOTIS-Will