ROBOTIS-GIT/OpenCR #244: turtlebot3_waffle example code failed connection with devices

fujimoto-shi, 2020-06-15:
I want to use a modified Turtlebot3 wheel with ROS2 dashing. The sketch of turtlebot3_waffle in the OpenCR example is I'm going to improve the base, but it doesn't work because of communication error.
Sketch is **Example -> Turtlebot3 ROS2 -> turtlebot3_waffle**

When I run the turtlebot3_bringup on Raspberry pi and start the robot, it fails to communicate with OpenCR.
ros2 launch turtlebot3_bringup robot.launch.py 

If you follow the OpenCR setup procedure described in the Turtlebot3 tutorial, Dynamixel's motors work fine.

> $ export OPENCR_PORT=/dev/ttyACM0
> $ export OPENCR_MODEL=waffle
> $ cd ~/opencr_update && . /update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

The following is an error message.

> ///////////////////////////////////////////////
> ros2 launch turtlebot3_bringup robot.launch.py 
> ///////////////////////////////////////////////
> ubuntu@ubuntu:~/opencr_update$ ros2 launch turtlebot3_bringup robot.launch.py 
> [INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2020-06-12-09-30-44-> 697008-ubuntu-3716
> [INFO] [launch]: Default logging verbosity is set to INFO
> urdf_file_name : turtlebot3_waffle_pi.urdf
> [INFO] [robot_state_publisher-1]: process started with pid [3726]
> [INFO] [hlds_laser_publisher-2]: process started with pid [3727]
> [INFO] [turtlebot3_ros-3]: process started with pid [3728]
> [robot_state_publisher-1] Initialize urdf model from file: /home/ubuntu/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_waffle_pi.urdf
> [robot_state_publisher-1] Parsing robot urdf xml string.
> [robot_state_publisher-1] Link base_link had 7 children
[robot_state_publisher-1] Link camera_link had 1 children
[robot_state_publisher-1] Link camera_rgb_frame had 1 children
[robot_state_publisher-1] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-1] Link caster_back_left_link had 0 children
[robot_state_publisher-1] Link caster_back_right_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] got segment base_footprint
[robot_state_publisher-1] got segment base_link
[robot_state_publisher-1] got segment base_scan
[robot_state_publisher-1] got segment camera_link
[robot_state_publisher-1] got segment camera_rgb_frame
[robot_state_publisher-1] got segment camera_rgb_optical_frame
[robot_state_publisher-1] got segment caster_back_left_link
[robot_state_publisher-1] got segment caster_back_right_link
[robot_state_publisher-1] got segment imu_link
[robot_state_publisher-1] got segment wheel_left_link
[robot_state_publisher-1] got segment wheel_right_link
[robot_state_publisher-1] Adding fixed segment from base_footprint to base_link
[robot_state_publisher-1] Adding fixed segment from base_link to camera_link
[robot_state_publisher-1] Adding fixed segment from camera_link to camera_rgb_frame
[robot_state_publisher-1] Adding fixed segment from camera_rgb_frame to camera_rgb_optical_frame
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_left_link
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_right_link
[robot_state_publisher-1] Adding fixed segment from base_link to imu_link
[robot_state_publisher-1] Adding fixed segment from base_link to base_scan
[robot_state_publisher-1] Adding moving segment from base_link to wheel_left_link
[robot_state_publisher-1] Adding moving segment from base_link to wheel_right_link
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[turtlebot3_ros-3] [ERROR] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-dashing-rcl-0.7.8/src/rcl/node.c:476
[ERROR] [turtlebot3_ros-3]: process has died [pid 3728, exit code -6, cmd '/home/ubuntu/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 __params:=/home/ubuntu/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/waffle_pi.yaml'].
`

fujimoto-shi, 2020-06-15:
I solved the problem. When I changed the version of Dynamixel2Arduino to 2.1, It works. Thank you.

closed 2020-06-15 by fujimoto-shi

finiel, 2020-06-30:
I also want to change wheel radius, but unfortunately, `fujimoto-shi`'s solution is not working for me. I need help.
My environment is Ubuntu 18.04, ROS2 dashing, Turtlebot3 waffle(with huge wheel), and have a similar error message.
I tried to upload Arduino `Examples for OpenCR Board/ TurtleBot3 ROS2/ turtlebob3_waffle` with `Dynamixel2Arduino v0.2.1~v0.4.4` and no one working properly(The tutorial opencr setup process is working). What should I do next?

``` bash
...
[turtlebot3_ros-4] [INFO] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-4] [INFO] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-4] [INFO] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-4] [INFO] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[turtlebot3_ros-4] [ERROR] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-4] [INFO] [turtlebot3_node]: Add Motors
[turtlebot3_ros-4] [INFO] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-4] [INFO] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-4] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-4]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-dashing-rcl-0.7.8/src/rcl/node.c:476
[ERROR] [turtlebot3_ros-4]: process has died [pid 8693, exit code -6, cmd '/home/kist/robotis/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 __params:=/home/kist/robotis/install/turtlebot3_bringup/share/turtlebot3_bringup/param/waffle.yaml cmd_vel:=cmd_velocity'].
```

Thanks for your help.