ROBOTIS-GIT/OpenCR #245: TURTLEBOT3_ROS2/turtlebot3_waffle doesn't connect with SBC

DaisukeUra, 2020-06-25:
Hi there,

I want to move tb3 with om.
So, I'm trying to edit the tb3 firmware for ros2.

I compiled the turtlebot3_waffle for ros2 in this repository and wrote it to OpenCR.
Then I connected to the RaspberryPi3B+ and OpenCR, and executed the following command.

`ros2 launch turtlebot3_bringup robot.launch.py`

However, the connection did not work.

But I try to burn the following binaries, and the ROS2 node was successfully connected and operated.
https://github.com/ROBOTIS-GIT/OpenCR-Binaries 
(I followed https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/#opencr-setup)

Is this binary not generated from this sample code?
Or, is there a ROS2 node for this firmware?

Best regards,

---

Errors I got is bellow.

```
[turtlebot3_ros-3] [INFO] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[turtlebot3_ros-3] [ERROR] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-dashing-rcl-0.7.8/src/rcl/node.c:476
[ERROR] [turtlebot3_ros-3]: process has died [pid 4624, exit code -6, cmd '/home/ubuntu/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 __params:=/home/ubuntu/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/waffle.yaml'].
```

ROBOTIS-Will, 2020-06-30:
Hi, in order to use a custom code for TurtleBot3 with OpenMANIPULATOR, please use DYNAMIXEL2Arduino library version 2.1. (#244)

DaisukeUra, 2020-06-30:
Thank you for your information!
I'll try it.

closed 2020-06-30 by DaisukeUra

DaisukeUra, 2020-07-01:
Can't compile the sources with DYNAMIXEL2Arduino `v0.2.1`.
So I use `v0.3.0`, it was compiled and connected with ros2 well.
Thank you for your support!

ROBOTIS-Will, 2020-07-01:
Thank you for your feedback.
We'll review the library and update it to support the latest version.