ROBOTIS-GIT/OpenCR #247: OpenCR uploading programs  successfully but nothing happens

Wolfman038, 2020-08-12:
I can upload sketches to the OpenCR through Arduino just fine.  The issue is nothing happens that is supposed to.  For example, when i upload the blink_LED example, it uploads to the board just fine, but the LEDs dont do anything.  Even the OM-X examples aren't working anymore i.e. chain_teaching.  I think its something with the board  but firmware recovery mode doesnt seem to help.

ROBOTIS-Will, 2020-08-27:
Unfortunately, I wasn't able to reproduce the issue you are experiencing.
You might have damaged OpenCR.
Have you also tried [Updating Bootloader](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#update-bootloader)?

tzf230201, 2020-10-23:
I also experienced the same thing some time ago, it solved by overwritten the old arduino-openCR library with the new arduino-openCR library on github.

ROBOTIS-Will, 2020-12-14:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-14 by ROBOTIS-Will