ROBOTIS-GIT/OpenCR #249: OpenCR usage with 4-6 cell batteries

giovanidiniz, 2020-10-06:
@OpusK 

Hey there!
 I have been working on a robot using the OpenCR board and I would like to use a larger battery on it.
The OpenCR is rated to accept 7-24V input either on the T-plug or on the barrel, if I understand the manual correctly. 
I decided to check the voltage on the TTL pins before plugging in my DXLs. 
When I plug in  a standard 3S battery on it (@11.1V) I read 7.4V on the plugs. 
If I plug in a 4S battery(@14.8V), I read the same 14.8V on the TTL pins. 
I'm now left wondering if If I can indeed power it and the DXLs with voltage higher than a standard 3S.
I appreciate any insights you may provide
Best

ROBOTIS-Will, 2020-10-07:
Hi @giovanidiniz,
The OpenCR does not regulate the input voltage to the DYNAMIXEL pins, but only use an FET to open and close the power source.
This feature allows users to use DYNAMIXEL with different operating voltages such as P series(24V) and XL-320(7.4V).
Therefore, in order to use XM430 with OpenCR, you should use 3S battery.
Thank you.

giovanidiniz, 2020-10-07:
I see, so how would I supply the XM430 with 12V using the OpenCR board?

Based on what you said, it would make sense to me if the board were to supply the 7.4V from a 2S LiPo not from a 3S? I am reading 7.4V from a 3S battery, though.

Thanks

ROBOTIS-Will, 2020-10-08:
HI @giovanidiniz ,
For most X series that runs off of 12V, you can use 3S LiPo battery something like [this](http://en.robotis.com/shop_en/item.php?it_id=903-0210-000).
If your 3S battery is reading 7.4V only, one of your battery cell is damaged. Nominal voltage for 3S battery is 11.1V
Thank you.

giovanidiniz, 2020-10-08:
Sorry for the misunderstanding.

My battery reads 11.1V on the connectors, so I know the input on the board is 11.1V

What I don’t understand is that of I provide an input of 11.1V to the OpenCR, only about 7.4V are provided to the TTL connectors. If, however, I provide an input of 14.8V I read the same 14.8 V on the TTL connectors.

ROBOTIS-Will, 2020-10-08:
By any change, have you modified the Arduino sketch based on the TurtleBot3 example?
The TurtleBot3 example will cut off the OpenCR FET between the power source and DYNAMIXEL ports once the voltage drops to 11.0V
Since there could be a small voltage drop across connectors and other ICs, your 11.1V may not able to turn on the FET.

giovanidiniz, 2020-10-08:
Aha! I was testing the board before flashing the firmware anything on it.

I’m running my own FW on it so glad to know I didn’t mess that up!

Thanks so much for the help

closed 2020-10-08 by giovanidiniz