ROBOTIS-GIT/OpenCR #259: Problems with Arduino CLI 0.14

pablogs9, 2020-12-14:
I have found a strange behavior when building OpenCR sketches with Arduino CLI:

## Steps to replicate

```bash
docker run -it --rm --net=host ubuntu:20.04
apt update
apt install curl lib32z1

# Installing Arduino CLI 0.14
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh 

# Installing OpenCR 1.0 Support
echo '''board_manager:
  additional_urls:
    - https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json''' > arduino-cli.yaml
arduino-cli core install OpenCR:OpenCR -v
arduino-cli core update-index
arduino-cli lib update-index

# Build and example
arduino-cli compile --fqbn OpenCR:OpenCR:OpenCR /root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/libraries/OpenCR/examples/01.\ Basics/a_Bare_Minimum/a_Bare_Minimum.ino
```

<details>

```bash
Using board 'OpenCR' from platform in folder: /root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16
Using core 'arduino' from platform in folder: /root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16
Detecting libraries used...
/root/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10607 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr/include -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/driver -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/usb_cdc -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/cores/arduino -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/sketch/a_Bare_Minimum.ino.cpp -o /dev/null
Generating function prototypes...
/root/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10607 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr/include -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/driver -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/usb_cdc -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/cores/arduino -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/sketch/a_Bare_Minimum.ino.cpp -o /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/preproc/ctags_target_for_gcc_minus_e.cpp
/root/.arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/root/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -std=gnu++11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10607 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr/include -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/driver -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/usb_cdc -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/cores/arduino -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/sketch/a_Bare_Minimum.ino.cpp -o /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/sketch/a_Bare_Minimum.ino.cpp.o
Compiling libraries...
Compiling core...
Using previously compiled file: /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/core/bsp/opencr/startup/startup_stm32f746xx.S.o
/root/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-gcc "	-c" -g -O2 -std=gnu11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10607 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr/include -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/driver -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/usb_cdc -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/cores/arduino -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR /root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr/bsp.c -o /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/core/bsp/opencr/bsp.c.o
/root/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-gcc "	-c" -g -O2 -std=gnu11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10607 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr/include -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/driver -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/hw/usb_cdc -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/ -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/cores/arduino -I/root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR /root/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.16/variants/OpenCR/bsp/opencr/stm32f7xx_it.c -o /tmp/arduino-sketch-347DF42F6FE3FCF015DC3A24294F2CD8/core/bsp/opencr/stm32f7xx_it.c.o
/root/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-gcc "	-c" -g -O2 -std=gnu11 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_OpenCR -mcpu=cortex-m7 -DF_CPU=2 [... cut ...]
arm-none-eabi-gcc: error: 	-c: No such file or directory
arm-none-eabi-gcc: error: 	-c: No such file or directory
arm-none-eabi-gcc: error: 	-c: No such file or directory
arm-none-eabi-gcc: error: 	-c: No such file or directory
arm-none-eabi-gcc: error: 	-c: No such file or directory
arm-none-eabi-gcc: error: 	-c: No such file or directory
arm-none-eabi-gcc: error: 	-c: No such file or directory
arm-none-eabi-gcc: error: 	-c: No such file or directory

Error during build: exit status 1
```
</details>

ROBOTIS-Will, 2020-12-14:
Hi @pablogs9 
Thanks for the PR. We'll review this and merge into the master soon.
Have a great day!

pablogs9, 2020-12-14:
Cool, thanks! Please close when you consider it solved.

pablogs9, 2021-01-12:
Hello, are there any updates on this issue? Are you planning to release changes in #260 to master?

ROBOTIS-Will, 2021-01-13:
@pablogs9 
Thank you for the friendly reminder. 
The develop branch will be merged into the master for the release within a couple week along with TurtleBot3 binary files.
Thank you for your patience.

ROBOTIS-Will, 2021-01-18:
@pablogs9 
Thank you for your patience.
The updated OpenCR library is released.
Please let us know if you have any difficulty.

closed 2021-01-18 by ROBOTIS-Will

pablogs9, 2021-01-18:
Thanks a lot!