ROBOTIS-GIT/OpenCR #263: Modify turtlebot_motor_driver

Valeria258, 2021-02-13:
I'm triying to modify this file https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/turtlebot3/turtlebot3_motor_driver.cpp in order  to add a new function to read current on the dynamixel servomotors but when I compile it i got an error saying that my function doesn't exist. It seems that is not using my modified files. Which files should I modify to solve this problem?

ROBOTIS-Will, 2021-02-15:
Hi @Valeria258,
Did you append your custom function to the header file where defines the Turtlebot3MotorDriver class?

TimothyHo584, 2021-03-11:
Can any one know that what is "VELOCITY_CONSTANT_VALUE" in turtlebot3_motor_driver.h come from?
I really confuse about the value 0.229 and 0.10472 come from.
Because I change the motor to MX-64, do I need to change it?

ROBOTIS-Will, 2021-03-11:
The constant `0.229` came from the [Velocity unit of DYNAMIXEL](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#goal-velocity104)
`0.10472` came from the calculation of angular velocity = 2 * pi / 60 = 0.10472
If you are going to use MX-64(2.0), the firmware that supports DYNAMIXEL Protocol 2.0, you don't need to modify these constants.

TimothyHo584, 2021-03-11:
> The constant `0.229` came from the [Velocity unit of DYNAMIXEL](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#goal-velocity104)
> `0.10472` came from the calculation of angular velocity = 2 * [...]

Thanks a lot!!I finally know these two constant means.

yun-goon, 2025-02-04:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-04 by yun-goon