ROBOTIS-GIT/OpenCR #265: recognizing board at second upload

theFabFor, 2021-03-02:
I'm hoping someone can tell me what I'm doing wrong.

I have recently purchased an openCR board and 2 Dynamixel RX-64 servos. While setting them up I'm getting similar messages from each of three methods I've tried to use. Once with a Dell XPS running Ubuntu 20.04, once with an older Dell Inspiron 540 running Ubuntu 12.04, and once with a Mac Mini, late 2012, running Mojave 10.14. 

Please see attached screenshots demonstrating the results on the Mac. I have had some success downloading to the device, the latest being the o_Find_Dynamixel program through Arduino (that successfully found the attached servo). The problems occur when trying to upload another program after the first. It seems as if the port changes (on the Mac it adds a "1" to the port after the serial number, on the Dell XPS the port disappears from the ports menu) then it's not possible to choose the "new" port. Any assistance would be greatly appreciated.

[image]
[image]

ROBOTIS-Will, 2021-03-03:
@theFabFor 
If you have left the Serial Monitor opened when uploading the Arduino sketch, previously opened port remains connected to the serial monitor.
Please close the serial monitor before uploading the sketch.

theFabFor, 2021-03-03:
Thanks for the advice to close the serial window: one less variable to be concerned with. However it doesn't resolve the problem, installing programs is still inconsistent. There are two new bits of information that may help find the issue:

1. - In the last 12 hours, on the Mac Mini and in Ubuntu 20.04, I have experienced both success and failure using the same steps in each instance (same programs: b_Blink_LED & o_Find_Dynamixel from examples/OpenCR/DynamixelWorkbench, same installation steps, ensure that serial window is closed). At different times both programs have been successfully installed and both have failed to install on the Mac. Blink has been successfully installed on Ubuntu, find-Dynamixel consistently failed. Please see attached images for details.

2. - On Ubuntu, port ttyACM1 disappears (ttyACM0 is assigned to a fingerprint sensor) from the Arduino port menu once board is put in DFU mode and further progress fails.

This is the error message:
Arduino: 1.8.13 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
	at processing.app.Editor.lambda$handleBurnBootloader$61(Editor.java:2379)
	at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.

Any advice on what steps to take next would be appreciated.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

theFabFor, 2021-03-03:
Thanks for the advice to close the serial window: one less variable to be
concerned with. However it doesn't resolve the problem, installing programs
is still inconsistent. There are two new bits of information that may help
find the issue:

1. - In the last 12 hours, on the Mac Mini and in Ubuntu 20.04, I have
experienced both success and failure using the same steps in each instance
(same programs: b_Blink_LED & o_Find_Dynamixel from
examples/OpenCR/DynamixelWorkbench, same installation steps, ensure that
serial window is closed). At different times both programs have been
successfully installed and both have failed to install on the Mac. Blink
has been successfully installed on Ubuntu, find-Dynamixel consistently
failed. Please see attached images for details.

2. - On Ubuntu, port ttyACM1 disappears (ttyACM0 is assigned to a
fingerprint sensor) from the Arduino port menu once board is put in DFU
mode and further progress fails.

This is the error message:
Arduino: 1.8.13 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

java.lang.NullPointerException
at
cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
at processing.app.Editor.lambda$handleBurnBootloader$61(Editor.java:2379)
at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.

Any advice on what steps to take next would be appreciated.

Randy Schack

ROBOTIS-Will, 2021-03-04:
For me, it still looks like a port related issue.
Are you running Linux on virtual machine?
When download fails, disconnecting OpenCR from the PC and connecting again resolves the issue?
How about downloading in [Recovery mode](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#recovery-mode)?
Disconnecting the fingerprint sensor and connecting OpenCR to ttyACM0 resolves the issue?

theFabFor, 2021-03-04:
Are you running Linux on virtual machine? *No. Ubuntu 20.04 running as the
sole operating system on Dell XPS 13*
When download fails, disconnecting OpenCR from the PC and connecting again
resolves the issue? *No.*
How about downloading in Recovery mode
<https://emanual.robotis.com/docs/en/parts/controller/opencr10/#recovery-mode>?
*As outlined in my last communication the port (ttyACM1), while still
appearing when doing LSUSB in the terminal indicates that the board has
changed to DFU mode, disappears from the Arduino port menu.*
Disconnecting the fingerprint sensor and connecting OpenCR to ttyACM0
resolves the issue? *The fingerprint sensor is built into the Dell. No
option to remove it.*

While I'm no expert I do have considerable time spent working with sbcs
(Arduino, Beaglebone Black, and Raspberry Pi). Unfortunately this board is
taking too much time just to get to a place where programs can be installed
reliably, if at all. I feel it's time to move on to another solution.

Randy

ROBOTIS-Will, 2021-03-05:
I'm sorry about the inconvenience you've experienced.
If you feel like the board is defected, please feel free to request RMA for inspection and repair.
The OpenCR board is under warranty for 1 year from the initial purchase date.
http://en.robotis.com/model/page.php?co_id=repaircenter
Thank you.

theFabFor, 2021-03-05:
At this point I’m not looking to repair the board. I am looking for a return authorization for the board purchased through Digital-Key (Robotis order #165739, Digi-Key #68182598-A) as well as the two RX-64 servos and 240mm cables ordered directly from Robotic (Robotis order #1-337408-5-4810-339331-165756).

Thank you,
Randy Schack

ROBOTIS-Will, 2021-03-08:
Clearly understood.
I hate to see you go, but hope you can find a better solution for your projects.
Thank you.

theFabFor, 2021-03-08:
Will,

Can you provide what steps are necessary to obtain an rma for return of the
two servos?

Thanks,
Randy Schack

ROBOTIS-Will, 2021-03-09:
Randy,

Unfortunately, I do not have detailed  the return / refund process.
I believe you can ask for the service where you purchased the product.
If you have purchased from the robotis.us site, please refer [here](https://www.robotis.us/shipping-returns-warranty/) for more details about returning items.
Thank you.

theFabFor, 2021-03-13:
Will,

Thanks for the response.

I noticed that Robotis has a policy of charging a 20% restocking fee. In
this particular instance, as we haven't yet come to a conclusion to what
the actual problem is, I have a proposal: please send me an RMA for
inspection and repair of the openCR board. At Robotis' discretion, if it is
found that I am at fault for the issues I'm having I will gladly pay the
restocking fee with my apology for wasting your time. Alternatively, if it
is discovered that there is a problem that requires the company's input I
would respectfully ask that the company waive the restocking fees for both
the board and the Dynamixel servos.

Randy Schack

ROBOTIS-Will, 2021-03-15:
Randy,

Unfortunately, I'm not the right person to make such deal.
Please consult with the RMA manager.
Thank you.

theFabFor, 2021-03-15:
Is this the correct address to contact that person?
http://en.robotis.com/model/page.php?co_id=repaircenter

ROBOTIS-Will, 2021-03-15:
The link you provided is not available for your region.
Please contact your local distributor at https://www.robotis.us/shipping-returns-warranty/
Thank you.

closed 2025-02-04 by yun-goon