ROBOTIS-GIT/OpenCR #269: Battery check logic

klues, 2021-03-16:
Hello, can someone please answer my questions to the battery check in [turtlebot3_diagnosis.cpp](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/turtlebot3/turtlebot3_diagnosis.cpp#L119):
* Is there some documentation on this?
* Why `voltage_ref` and `voltage_ref_warn` are both `11.0` is there any reasoning behind it? Do you have a datasheet of the battery?
* What's the purpose of these `battery_valtage_raw > 5.0` checks? Shouldn't this always be the case?
* What does `setPowerOff()` and `setPowerOn()` do? On/off the power of the dynamixel motors?
* Is there any low voltage protection for the battery or is it already included in the battery?

ROBOTIS-Will, 2021-03-17:
Hi @klues 

- The TurtleBot3 firmware sketch is provided as an open source without a documentation. 
- The reference voltage is set to 11.0V because TurtleBot3 uses 3S (11.1V nominal) LiPo battery, and this value is somewhat configured with a safety margin.  
  `voltage_ref_warn` defines the minimum voltage to set the alarm, while `voltage_ref` disengages the power to DYNAMIXEL with `setPowerOff()` when low voltage.

- If you need to get the datasheet of the battery and PCM, please contact at contactus2@robotis.com.
- I'm not sure what the 5.0V check do in TurtleBot3, but it seems like it was implemented for other platform with lower voltage.
- The battery is included with a PCM to protect from overcharge, discharge, excessive current protection.  
  However, it is recommended to take extra care not to over discharge the LiPo battery as it may cause damage on the cell.

Thanks!

klues, 2021-03-17:
Thanks for your quick answer! 👍 
I've contacted contactus2@robotis.com regarding the datasheet.

closed 2021-03-17 by klues