ROBOTIS-GIT/OpenCR #270: OpenCR rs485 port

chrswp, 2021-03-23:
Hello...
My name is Chris from Indonesia.
I would like to ask is it possible to use RS 485 port in openCR 1.0 board to control another device beside dynamixel. I have BLDC motor controller need to be accessed from OpenCR thru RS485, but my engineer there is some difficulties due to port in OpenCR is mod for dynamixel only.
Please advice.

Thanks
Chris WP

ROBOTIS-Will, 2021-03-23:
hi Chris,
RS-485 is one of standard communication layers that supports various devices, however, each device may need its own driver to control them.
For example, you can use RS-485 D+/D- for other devices, but OpenCR Arduino library only provides RS-485 for DYNAMIXEL.
In order to use other devices, you will need to add/create the library to control the device you want to use.
https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries
Thanks.

chrswp, 2021-03-23:
Hi Will, thanks for your reply. 
My BLDC motor driver is using common RS485 modbus protocol, is it still need library to communicate from openCR?

ROBOTIS-Will, 2021-03-23:
OpenCR library supports RS485 DYNAMIXEL protocol, so you should implement the modbus protocol library to use your BLDC motor.
Thanks.

yun-goon, 2025-02-04:
This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it.

closed 2025-02-04 by yun-goon