ROBOTIS-GIT/OpenCR #272: CheckSum does not match.

jhbirdchoi, 2021-04-08:
Hi

I am using the noetic version of ros1.

If I use an image that has already been built, there is no problem.

$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=burger_noetic
$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 
$ tar -xvf opencr_update.tar.bz2 
(Fw version : 1.2.6)

But when I build and use opencr on arduino, I have a problem.
The following error log is displayed.
"Creation of publisher Failed : Checksum does not match"

Which version of opencr should I use? (at Arduino board manager)

thanks

ROBOTIS-Will, 2021-04-08:
Hi @jhbirdchoi 
I'm sorry about the confusion.
Since Noetic, battery status message has been updated and requires a new library.
This fix has been applied to the Arduino example, but need to uncomment the `NOETIC_SUPPORT` definition in `turtlebot3_core_config.h` file.

1. Open the TurtleBot3 firmware. Depending on your platform, please select the correct firmware.
  Burger : File > Examples > turtlebot3 > turtlebot3_burger > turtlebot3_core
  Waffle/Waffle Pi : File > Examples > turtlebot3 > turtlebot3_waffle > turtlebot3_core
2. Open the turtlebot3_core_config.h and uncomment the NOETIC_SUPPORT defintion in the line 21.
3. Connect OpenCR to the PC and Select OpenCR > OpenCR Board from Tools > Board menu.
4. Select the OpenCR connected USB port from Tools > Port menu.
5. Upload the TurtleBot3 firmware sketch with Ctrl + U or the upload icon.

Above information is described in https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup `Noetic` document.
The content is folded by default and can be expanded by clicking on the expandable note.
[image]

jhbirdchoi, 2021-04-08:
Hi Will son

It works well!!!
Thank you for your help.

Best Regards.!

closed 2021-04-08 by jhbirdchoi