ROBOTIS-GIT/OpenCR #273: OpenCR board does not detect Dynamixels

mmorales45, 2021-04-08:
Remote PC OS: Ubuntu 16.04
OpenCR: 1.0
Dynamixel: XM430-W350T x5

My team is working with a Openmanipulator-X. The arm used to be able to be controlled with an OpenCR board in the past but now I keep having issues trying to control it. When the arm is connected to power, the torque is not enabled. 

I ran the code below and got that issue. 
[image]
[image]

From here, I bought a U2D2 and updated the firmware on each individual dynamixel and then assigned the correct ID for each actuator using the Dynamixel Wizard 2.0. From there, I scanned for each dynamixel again, on the U2D2, and each was recognized. 
I then connected everything back to the OpenCR board and tried running the Dynamixel Wizard against and none appear. 

Is there something wrong with the OpenCR board or am I missing something?

Thank you!

ROBOTIS-Will, 2021-04-09:
Hi @mmorales45 ,

Unlike U2D2, OpenCR requires a firmware that can communicate with a specific hardware example.
In case of OpenMANIPULATOR-X example, you can use `usb_to_dxl` Arduino example firmware to communicate with DYNAMIXEL.
Please refer to below eManual to upload the firmware to OpenCR.
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#upload-source-code
Thanks!

mmorales45, 2021-04-11:
Thank you for the reply, I tried doing as you suggested and it went through successfully but the issue still persists. The picture below is the result of running the usb_to_dxl example.
[image]

All the Dynamixels are detected on the Dynamixel Wizard 2.0 when using a U2D2 but still fail to be detected on the OpenCR board. 
This next picture is the Dynamixels being recgonized by the U2D2.
[image]
The next picture is the Dynamixels not being found when using an OpenCR board. 
[image]
[image]

We have updated the bootloader already with no success in resolving our problem. What should our next step be?

Thank you.

ROBOTIS-Will, 2021-04-12:
@mmorales45 
Could you describe how power source is connected to OpenCR?
After switching on and launching the controller node, do you see any LED blink on each DYNAMIXEL?

mmorales45, 2021-04-13:
I connected the power source and other wires the same way as in the manual. 
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#communication-converter

A few things to note, when I do turn on the OpenCR board, sometimes the LEDs on the Dynamixels Flash red once but most of the time they do not blink at all. Even when the Dynamixels LEDs blink and I try to launch the controller, the error still persists where none of the Dynamixels are detected and the torque does not turn on.

ROBOTIS-Will, 2021-04-14:
@mmorales45 
What do you see when you enter the command below? Do you see `ttyACM0` or anything else like ttyACM1?
```bash
$ ls /dev/ttyACM*
```
You can use U2D2 and DYNAMIXEL Wizard 2.0 as a packet sniffer.
Connect as below and open [DYNAMIXEL Wizard 2.0 Packet](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#packet) window. 
[image]

Set the baudrate to 1Mbps and open the ttyUSB0 port of U2D2.
Then power up the OpenCR and launch the OpenMANIPULATOR controller node.
If everything goes well, you should see some incoming packets that initialize DYNAMIXEL.

ROBOTIS-Will, 2021-04-14:
What do you get in return when you enter below command on your terminal?
```bash
$ ls /dev/ttyACM*
```
Do you see anything else than `ttyACM0`?

Inconsistency of LED blink when powering on can be caused by damaged cable or loosen connectors.
Please check if all cables and connectors are firmly connected.

You can try sniffing the packet transmitted from OpenCR using U2D2 and Wizard 2.0
[image]
Using the [Packet feature of Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#packet), set baudrate to 1Mbps and open ttyUSB0 port where U2D2 is connected.
Then power up the OpenCR and launch the OpenMANIPULATOR controller.
You should be able to receive DYNAMIXEL Protocol 2.0 packets sent from OpenCR or status packet from each DYNAMIXEL.
Carefully examine each packet, especially, check if you can get status return packet from DYNAMIXEL ID 11 ~ 15.

yun-goon, 2025-02-04:
This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it.

closed 2025-02-04 by yun-goon