ROBOTIS-GIT/OpenCR #275: OpenCR Tools Error

Panda3047, 2021-04-15:
hello,

I am currently trying to do the steps for the OpenCR Hardware Setup for the OpenMANIPULATOR-X for the OpenCR board on Ubuntu 16.04. But when I run the first two "wget" links on the terminal I encounter a 404-page not found error. I then went to the websites and noticed that the page doesn't exist for the first two but the third one works fine. Is there a different link where these tools can be found? Because I noticed ROBOTIS-WILL changed a few things a few days ago about the packets.

[image]
[image]
[image]
[image]

ROBOTIS-Will, 2021-04-20:
Hi @Panda3047 
Thank you for reporting the missing files.
Since the eManual improperly connected to my personal repository, I'll fix this instruction.

In the meantime, please use Arduino IDE to upload the example from your PC.
You can find the installation instruction from the page below.
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#arduino-ide
Once the installation is completed, you can find the example from examples > OpenCR > 10. Etc > usb_to_dxl

Thank you.

yun-goon, 2025-02-04:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-04 by yun-goon