ROBOTIS-GIT/OpenCR #276: Is it possible give the power input 19v 4.5A on Open CR?

fuad99, 2021-04-26:
i have power issue with Open CR and Dynamixel, every Dynamixel is on, the power on Open CR is dropped to 8v and make Open CR is resetting. Any Soultion? 
Thanks very much!

ROBOTIS-Will, 2021-04-27:
**Please be aware that OpenCR does not regulate the input voltage to DYNAMIXEL port.**
If you connect 19V to the input port, DYNAMIXEL will receive the same voltage which may damage the product if you are using 12V models.
Depending on the number of DYNAMIXEL connected to the OpenCR and the power source you use, voltage drop may occur.
Please try connecting the proper power source directly to DYNAMIXEL to avoid voltage drop from high current required application.
Thank you.

yun-goon, 2025-02-04:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-04 by yun-goon