ROBOTIS-GIT/OpenCR #279: Replace Raspberry3b

Daxturtle, 2021-06-03:
Hi, i'm new to OpenCR. I have installed raspbian with ros kinetic in raspberry3b+ and i work with turtlebot3 burger. 
I'd like to know if i reinstall ubuntu 18.04 with ros noetic in raspberry I have to change something in OpenCR or work directly? 
Thanks

ROBOTIS-Will, 2021-06-04:
Hi @Daxturtle 
There is a change of message structure from ROS1 Noetic and you should update the OpenCR firmware for Noetic as described in the eManual.
Otherwise, you'll get an error like this(https://github.com/ROBOTIS-GIT/turtlebot3/issues/692#issuecomment-754477733)
Thank you.

Daxturtle, 2021-06-07:
Thanks for your reply!

yun-goon, 2025-02-04:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-04 by yun-goon