ROBOTIS-GIT/OpenCR #280: OpenCR uart7 not working.

Indruino-hieu, 2021-06-16:
Hi Team! 

I'm beginer work with opencr board. When i do config UART7 on Opencr board with chip STM32F746ZGTx, it's not received data. 
It's only transmit data.
My source:
else if (huart->Instance == UART7) // // UART_NUM_7
  {
    RCC_PeriphCLKInitStruct.PeriphClockSelection = RCC_PERIPHCLK_UART7;
    RCC_PeriphCLKInitStruct.Uart7ClockSelection = RCC_UART7CLKSOURCE_SYSCLK;
    HAL_RCCEx_PeriphCLKConfig(&RCC_PeriphCLKInitStruct);

    /* Peripheral clock enable */
    //__HAL_RCC_DMA2_CLK_ENABLE();
    __HAL_RCC_UART7_CLK_ENABLE();

    GPIO_InitStruct.Pin = GPIO_PIN_6;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF8_UART7;
    HAL_GPIO_Init(GPIOF, &GPIO_InitStruct);

    GPIO_InitStruct.Pin = GPIO_PIN_7;
    GPIO_InitStruct.Alternate = GPIO_AF8_UART7;
    HAL_GPIO_Init(GPIOF, &GPIO_InitStruct);

    /* Peripheral interrupt init */
    HAL_NVIC_SetPriority(UART7_IRQn, 0, 0);
    HAL_NVIC_EnableIRQ(UART7_IRQn);
  }

Could you help me fix it, thanks so much! 
Thanks to all.

ROBOTIS-Will, 2021-06-16:
UART7_TX and RX pins are assigned for ADC4 & 5 for Arduino shield.
You may revise these settings from the variant.cpp below.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/variant.cpp#L61

There will be some additional works to be done such as adding TX/RX handlers and creating UARTClass for the serial port in variant.h
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/variants/OpenCR/variant.h

You'll also need to remove ADC configurations for A4 and A5 pins.

Please understand that we do not provide a thorough support on customizing the source code.
Thank you.

Indruino-hieu, 2021-06-17:
Hi Will, 

Thanks to you reply forme!

I know that, your team do not support customizing the source code.
I create this issue to verify UART7 can work when i config disable ADC feature in pin F6 and F7. 
Please tell me know any bug when i do it! 

Thanks you so much!

ROBOTIS-Will, 2021-06-17:
@Indruino-hieu 
You might want to utilize the community where many users hang out and there might be other developers who'd be interested in what you are doing.
https://community.robotis.us/

Since this is not an issue of the product or the source code, I'll close this thread.
Thank you.

closed 2021-06-17 by ROBOTIS-Will