ROBOTIS-GIT/OpenCR #282: DXL UART

dabenpb, 2021-08-22:
Is it possible to access the UART3 directly as a standard UART prior to the TTL/485 duplexer?  This would be without using any of the dyna libraries.  I have a need for 4 uarts.

ROBOTIS-Will, 2021-08-23:
Hi @dabenpb 
The USART3 is connected to the DYNAMIXEL port via RS485 or TTL IC.
Since these pins are not exposed as GPIO, you may not be able to access them.
If you are using Arduino IDE, have you considered using a softwareserial?

dabenpb, 2021-08-23:
I was under the impression that SoftwareSerial was not supported based on a previous question I asked.

ROBOTIS-Will, 2021-08-24:
@dabenpb 
Sorry about the confusion. What I meant in the previous inquiry was that we have no plan for library update.
You can use SoftwareSerial library from Arduino as OpenCR supports [DYNAMIXEL Shield](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/) which uses softwareserial for built in serial port.
Thank you.

sunghowoo, 2025-02-24:
this issue has been inactive for a long time, we will proceed with closing it. If you have any further concerns, please feel free to reopen the issue or create a new one.

closed 2025-02-24 by sunghowoo