ROBOTIS-GIT/OpenCR #283: XC430-W240-T - Motor won't move

AndrewWijaya888, 2021-08-30:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  Last Release

2. Which programming language/tool do you use?
  Arduino IDE 1.8.8

3. Which operating system do you use?
  Windows 10

3. Which USB serial converter do you use?
  OpenCR

4. Which DYNAMIXEL do you use?
  XC430-W240-T

5. Have you searched the issue from the closed issue threads?
  Yes

6. Please describe the issue in detail
I want to use XC430-W240-T with the default firmware for OpenCR but I didn't move.

First, I used the default XL430-W250. I followed the instruction to set up the OpenCR from this link https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup, after this I was able to to the test by pressing SW1 and SW2 and both motor moves.

However, when I swap both motors with two XC430-W240-T and run the test, it won' move. I tried connecting one XL and one XC, in this case, only the XL works.

Any recommendations and solutions are welcome!

7. How can we reproduce the issue?
  Simply follow the setup guide for OpenCR, then connect XC430-W240-T motors.

ROBOTIS-Will, 2021-08-30:
Hi @AndrewWijaya888 
Before you can use any DYNAMIXEL for TurtleBot3, the DYNAMIXEL has to be configured first.
https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3

AndrewWijaya888, 2021-09-01:
Hi @ROBOTIS-Will 
Thank you for your help.

I have followed the manual you sent me, but I encountered a problem as you can see below.
[image]

I tried inputting 3 and 4 for testing, but it shows "Test Motor Left...    dxl motor ID:1 not found". How do I solve this?

For clarity, here is my connection setup for the OpenCR and Dynamixel.
[image]

ROBOTIS-Will, 2021-09-06:
Hi @AndrewWijaya888 

When setting up the DYNAMIXEL, please connect only one module at a time.
Connecting multiple DYNAMIXEL modules will cause communication collision.

AndrewWijaya888, 2021-09-08:
Hi @ROBOTIS-Will,

This solves the problem, the motor is working now. Thank a lot for your help!

closed 2021-09-08 by AndrewWijaya888