ROBOTIS-GIT/OpenCR #286: How do I flash ROS library changes to OpenCR?

royito55, 2021-10-05:
I am trying to get rid of the low voltage warning buzzer. I realized that the odometry doesn't work after this.

I followed these instructions to find where I need to deactivate the alarm:

https://answers.ros.org/question/296145/turtlebot3-low-battery-beeping/?comment=387904#post-id-387904

But I don't know the actual steps to reupload the opencr packages? The manual simply says

How to modify ROS library code
You can modify ROS library in below paths.
Your/Arduino/Board/Package/Directory/OpenCR/hardware/OpenCR/Version/libraries/turtlebot3
Your/Arduino/Board/Package/Directory/OpenCR/hardware/OpenCR/Version/libraries/turtlebot3_ros_lib
If you want to modify it in the sketch folder, move the above two folders to the “skechbook/libraries” folder.

And I've found that path, but I can't open the .cpp in the Arduino IDE.

How do I flash these changes to the OpenCR?

ROBOTIS-Will, 2021-10-15:
Hi @royito55 
In order to modify the firmware in the OpenCR board manager library, you could use an editor like VSCode to open the cpp file.
Once the modification is done, the updated firmware can be uploaed to the OpenCR via Arduino IDE by selecting the example you modified.
You can find instruction in the eManual as below.
Please note that the OpenCR board manager doesn't support ARM based CPU, therefore, in order to build and upload the firmware, you should use your laptop or PC.
[image]
Thanks!

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, we will proceed with closing it. If you have any further concerns, please feel free to reopen the issue or create a new one.

closed 2025-02-24 by sunghowoo